klein0r/ioBroker.octoprint #1: Kann nicht mit octopi verbinden

eagleye2, 2018-07-10:
Zuerst einmal danke für den Adapter
Ich kann mich leider nicht mit octopi verbinden. Habe die richtige API eingetragen. Trotzdem bleibt das Licht Gelb. Kann es sein dass es mit den Access Control zusammenhängt?
Habe ein Username und Passwort, dass ich nirgends eintragen kann.
Danke vorab.

RoboCop0815, 2018-09-10:
Moin Moin,
habe das gleiche Problem... Anzeige bleibt gelb...
habe auch User und PW 
Grüße Robbie

ZeroCool80, 2018-09-11:
Servus,
habe auch keine Verbindung!

Gruß

mrMuppet, 2018-11-06:
bei mir leider auch.

Moormerlander, 2018-11-26:
Wird dieser Adapter nicht mehr gepflegt?

kxn25, 2018-12-20:
Sieht nicht aus als würde weiter daran gearbeitet!
Also ich komm auf den Printer mit Port 80.
Aber mein Iobroker kriegt immer LOG mit der Meldung null, das nerft...

Camel2k, 2019-02-21:
Gibt es eine Lösung mittlerweile?

Moormerlander, 2019-02-21:
Offensichtlich pflegt das niemand mehr. Ich habe es über FHEM gelöst, da ich meinen verbliebenen Homematic-Kram noch damit betreibe und nicht extra ne CCU für das bisschen aufsetzen will.  So mache ich das auch mit Octoprint. Über den FHEM-Adapter funzt Octoprint super und ich kann alle möglichen Stati (Readings) abfragen und in meine Scripts einbauen.

klein0r, 2020-08-25:
Der Adapter wurde in den letzten Versionen komplett überarbeitet.
Sollte alles behoben sein 👍

closed 2020-08-25 by klein0r

npeter, 2020-11-26:
Verbindung mit API nicht erfolgreich:
Log...
octoprint.0 | 2020-11-26 01:19:10.148 | info | (29431) OctoPrint API not connected
octoprint.0 | 2020-11-26 01:19:10.147 | info | (29431) getaddrinfo ENOTFOUND http
octoprint.0 | 2020-11-26 01:19:10.106 | info | (29431) starting. Version 1.0.1 in /opt/iobroker/node_modules/iobroker.octoprint, node: v12.18.4, js-controller: 3.1.6

Hätte eigentlich erwartet Account-Daten konfigurieren zu müssen.
Hat jemand eine Idee?

npeter, 2020-11-29:
Problem gelöst. Hatte URL statt IP konfiguriert. Adapter funktioniert!
Danke für den Adapter!!!

klein0r, 2020-11-29:
@npeter Sehr gut!

*Für die Zukunft aber bitte einen neuen Issue öffnen, und nicht an einen geschlossenen anhängen, welcher über ein Jahr alt ist.*