klein0r/ioBroker.octoprint #2: Error Meldung wenn Raspberry PI aus ist.

DenwoT, 2018-07-23:
Hallo, erst einmal recht vielen dank für den Adapter. Macht das Drucken um einiges einfacher.

Wäre es möglich, das man ERROR-Meldungen deaktivieren kann? 

Im Moment schaltet sich der Drucker ab einer bestimmte Uhrzeit aus bzw. ich kill den Strom. Dann ist aber auch nicht mehr mein Raspberry pi zu erreichen und somit Spamt er im Log das der Host offline ist.
Besteht die Möglichkeit das man es irgendwo deaktivieren kann? Bzw diese Message heraus nehmen kann?

`octoprint.0 | 2018-07-23 09:15:50.105 | error | Error: connect EHOSTUNREACH 192.168.*.*`

turbolift, 2018-12-23:
Bei mir ist es ähnlich:

Ich habe den Drucker grundsätzlich ausgeschaltet. Im Moment wird das Log im Minutentakt gefüllt.

```
octoprint.0 | 2018-12-23 11:37:15.394 | error | Status Code: 409 / Content: Printer is not operational
-- | -- | -- | --
octoprint.0 | 2018-12-23 11:37:14.604 | info | sending request to http://192.168.178.44:80/api/job with data: null
octoprint.0 | 2018-12-23 11:37:14.604 | info | sending request to http://192.168.178.44:80/api/printer with data: null
octoprint.0 | 2018-12-23 11:37:14.603 | info | sending request to http://192.168.178.44:80/api/connection with data: null
octoprint.0 | 2018-12-23 11:37:14.602 | info | sending request to http://192.168.178.44:80/api/version with data: null

```

resper, 2019-01-08:
Gleiches Problem bei mir. Schalte den PI vom 3D Drucker öfters stromlos, dann spamt mir Octoprint die Log Datei voll.

turbolift, 2019-01-08:
Ich habe aus diesem Grund das Projekt geforked.

p3pp0, 2019-01-13:
@turbolift Danke für deinen Fork. Leider funktioniert er bei mir nicht (installiert normal über Adapter-Custom-URL)... Error: connect EHOSTUNREACH 192.168.1.9:80 wird weiterhin regelmäßig angezeigt

resper, 2019-01-20:
Habe jetzt auch noch mal einen Fork erstellt und die Error Meldung auf das DEBUG Level gesetzt. Zu finden unter: https://github.com/resper/ioBroker.octoprint

p3pp0, 2019-01-26:
Danke @resper

turbolift, 2019-01-27:
Ich habe vergessen zu erwähnen, dass ich das LogLevel auf WARN gestellt habe um diese los zu werden.

[image]

moeff, 2019-11-19:
Meine Lösung ist:

nur wenn der OctoPrint Rechner (bei mir ein Raspi läuft) den Adapter zu starten. Ich starte diesen per Steckdose und fahre ihn per ssh vom iobroker per Skript runter. In diesem Ablauf startet auch der Adapter und beendet sich dementsprechend auch wieder ;)

klein0r, 2020-08-25:
In der aktuellsten Version kommt nur noch eine Info-Meldung wenn OctoPrint nicht erreichbar ist.

closed 2020-08-25 by klein0r