klein0r/ioBroker.octoprint #10: connected nicht mit default port 5000

tis-cs, 2019-12-27:
Da ich lang gesucht habe, wollte ich nur ne Lösung hier lassen.
Der eingestellte default pot 5000 musste bei mir mit Adapter 0.0.2 und Octoprint Version 1.3.12 auf Port 80 gestellt werden, da sonst keine Verbindung möglich war.

worksasdesigned, 2020-02-11:
danke! Das Problem hatte ich auch Version 1.3.9

kniwweli, 2020-04-09:
tis-cs funktioniert er noch bei dier?

tis-cs, 2020-04-09:
@kniwweli kann ich leider keine Aussage mehr zu treffen. Bin mittlerweile auf einen nativen Betrieb auf nem Pi4 umgestiegen. Sorry

kniwweli, 2020-04-09:
Nativer betrieb ?

Cannivy Luc

www.whatsinthebox.lu

tis-cs, 2020-04-09:
> Nativer betrieb ?

Sorry, vorhin nur zwischen den Zeilen gelassen. Iobroker läuft nativ und nicht mehr im Container. Dass es hier um den octoprint Adapter geht, hab ich vollständig überlesen. 

Bei mir läuft es noch in Version 0.0.2.

kniwweli, 2020-04-09:
Oki welchen port hast du denn bei mir wird er nicht grün wo finde ich den port

Cannivy Luc

www.whatsinthebox.lu

tis-cs, 2020-04-10:
@kniwweli ich habe im Adapter Port 80 und den API Key eingetragen

klein0r, 2020-04-11:
Genau, ich glaube in den neueren Versionen von OctoPrint läuft die API auf Port 80. Werde ich auch in den zukünftigen Releases berücksichtigen

JojoS62, 2020-05-31:
ich habe eine Multiprinter config nach dieser Bauanleitung erstellt:
http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers
das funktioniert auch, es muss nur '--host=$HOST' in den /etc/default/octoprint scripts weggelassen werden. Dann sind verschiedene Printer über die unterschiedlichen Ports erreichbar. Es müssen natürlich auch im iobroker mehrere octoprint Instanzen angelegt werden.

closed 2020-08-25 by klein0r