klein0r/ioBroker.octoprint #12: Please add you adapter to our repository

ioBrokerChecker, 2020-02-25:
We saw your adapter and if it is ready please request that it is added to the repository. With this it becomes available for all users. Please consult https://github.com/ioBroker/ioBroker.repositories#add-a-new-adapter-to-the-latest-repository and the following sections for more details.
If you need help, contact us in the forum https://forum.iobroker.net/ or in the developer chat https://github.com/iobroker-community-adapters/info .

Thank you for your support.

Mr-Markus, 2020-07-26:
@klein0r 
Kannst du bitte den Adapter hinzufügen dass er auch ohne Katze im ioBroker zu installieren ist?

Würde gerne ein Skript damit bauen, welches den Drucker bei Überhitzung abschaltet.

Danke

klein0r, 2020-07-26:
Der Adapter ist bereits im Latest-Repository verfügbar. Für Stable muss ich noch etwas testen und die letzten kleinen Fehler fixen. Also bitte nicht über die OctoCat installieren, sondern einfach aus dem latest-Repository ziehen.

closed 2020-07-26 by klein0r