klein0r/ioBroker.octoprint #18: Send „Printer Command“ not working

rnc11, 2020-10-24:
It seems that the possibility to send commands to OctoPrint is not working. 

I've tried several M or G commands without any effect.

Unfortunately there is no documentation how the commands need to formatted. But I would expect that an Value of "G28" in octoprint.0\command\printer should trigger the command.
I always get an "(23147) printer command not allowed: G28" back in the log.

**Versions:**  
 - Adapter version: 0.0.2
 - JS-Controller version: 3.1.6
 - Node version: v10.22.0
 - Operating system: linux (Raspberry PI 4)

klein0r, 2020-10-29:
You mean ".command.custom"? This is documented here: https://docs.octoprint.org/en/master/api/printer.html#send-an-arbitrary-command-to-the-printer

rnc11, 2020-10-29:
Hi,

 

I mean the “format” of the value that is needed here:

 

I have tried several values (with []{} “”…) but all without any effect.

I always get an error back in the ioBroker log: (13593) printer command not allowed: xyz

 

Could you please provide one running example of an G or M command that works (using the great ioBroker.octoprint adapter)?

 

regards

klein0r, 2020-10-29:
Just tested `G28` and everything worked perfectly.

[image]

Please provide more information about the tested command and the issue. I think you are using the wrong field for the command (as already mentioned in the last comment...).

closed 2020-10-30 by klein0r