klein0r/ioBroker.octoprint #19: octoprint.0.printjob.progress.completion wrong

p3pp0, 2020-10-24:
Hey Matthias,

leider ist die Prozentzahl unter octoprint.0.printjob.progress.completion seit dem letzten Octoprint-Update dauerhaft falsch... :-(

Wenn du das fixen könntest wäre genial, danke dir!

klein0r, 2020-10-26:
Kannst Du genauer definieren, was "falsch" bedeutet? Ungenau? Oder immer 0?

p3pp0, 2020-10-26:
Bitte entschuldige, das war keine gute Angabe von mir...

Der Prozentwert ist immer ganz deutlich zu niedrig bzw. entspricht nicht dem Prozentwert aus der Octoprint GUI. So, als hätte er teils mehrere Stunden "Delay".

Es liegt aber definitiv nicht am Refresh-Intervall - der Wert wird regelmäßig aktualisiert..

Bitte lass mich wissen wenn ich noch irgendwie weiterhelfen kann

klein0r, 2020-10-29:
In welcher Version hattest Du das Verhalten denn? In 0.0.6 ist bei mir alles korrekt berechnet.

Leviathan09, 2020-10-30:
Ich habe das Problem auch, benutze die Version 0.0.6
Habe dir mal einen Screenshot angehängt. Octoprint GUI 23%, bei den Datenpunkten des Adapters 17%
[image]

p3pp0, 2020-10-30:
@klein0r bei mir exakt genauso wie bei @Leviathan09 
Lass mich gerne wissen wenn ich noch irgendwelche Infos liefern kann

klein0r, 2020-10-30:
Habt ihr noch irgendwelche Plugins installiert, welche den Fortschritt und Restzeit genauer berechnen? Über die API wird als Fortschritt einfach nur die aktuelle Position in der gcode-Datei vs Gesamtgröße gerechnet.

Ilovegym66, 2020-10-30:
Hi, bei mir stimmt die Anzeige vom Adapter mit dem von Octoprint überein. Ich habe keine weiteren Plugins disbezüglich installiert. 
Vielleicht aktualisiert bei euch der Browser nicht schnell genug?

Leviathan09, 2020-10-30:
Ich habe ein Plugin installiert was mir die Zeit genauer berechnet
Aber weiß nicht ob das die Prozentangabe beeinflusst

klein0r, 2020-10-30:
@Leviathan09 Aber klar, darum geht es ja. Ohne die Plugins dauern z.B. die ersten Prozente super lange, weil z.B. eine Bodenplatte gedruckt wird. Da dauert es Stunden bis das erste Prozent fertig ist (was ja mathematisch keinen Sinn ergibt). Solche Plugins optimieren das dann. Ich mach hier zu.

closed 2020-10-30 by klein0r

p3pp0, 2020-11-01:
@klein0r : die Ursache ist (bei mir) das Plugin Print-Time-Genius.. deaktiviert man dieses, stimmen die Prozente wieder überein.
Die errechnete Endzeit ist halt dann wieder quatsch - die durch das Plugin-Verbesserte Endzeit wurde in iobroker auch wieder richtig angezeigt :-)

klein0r, 2020-11-02:
@p3pp0 Ich muss mal schauen, ob das Plugin auch die Daten per API ausgibt. Falls das Fall ist, kann ich stattdessen die Daten nehmen

p3pp0, 2020-11-03:
@klein0r das wäre klasse! Ohne Plugin sind nämlich die Prozent-Zahlen und auch die berechnete Endzeit völlige Grütze... mit Plugin  stimmt zumindest Rest/Endzeit ;-) Komisch, dass das Problem erst nach einem Update aufgetaucht ist..

klein0r, 2020-11-04:
@p3pp0 Mh, ich nehme nach wie vor die gleichen Zahlen aus der API. Ich kann nur nehmen, was ich bekomme.

p3pp0, 2020-11-15:
@klein0r  @Leviathan09 
Meine Lösung:

[image]