klein0r/ioBroker.octoprint #20: Falls nicht schon bekannt, bitte obj.common.type Objekte an neue IOBroker Anforderungen anpassen :)

Jarvid, 2020-10-26:
Im Log gibt es bereits Warnungen, in zukünftigen iobroker Versionen würde der Adapter wohl leider nicht mehr funktionieren.

octoprint.0 | 2020-10-26 21:18:52.867 | warn | (12331) This object will not be created in future versions. Please report this to the developer.
-- | -- | -- | --
octoprint.0 | 2020-10-26 21:18:52.867 | warn | (12331) Object octoprint.0.command.jog.z is invalid: obj.common.type has an invalid value (numeric) but has to be one of number, string, boolean, array, object, mixed, file, json
octoprint.0 | 2020-10-26 21:18:52.866 | warn | (12331) This object will not be created in future versions. Please report this to the developer.
octoprint.0 | 2020-10-26 21:18:52.865 | warn | (12331) Object octoprint.0.command.jog.y is invalid: obj.common.type has an invalid value (numeric) but has to be one of number, string, boolean, array, object, mixed, file, json
octoprint.0 | 2020-10-26 21:18:52.864 | warn | (12331) This object will not be created in future versions. Please report this to the developer.
octoprint.0 | 2020-10-26 21:18:52.863 | warn | (12331) Object octoprint.0.command.jog.x is invalid: obj.common.type has an invalid value (numeric) but has to be one of number, string, boolean, array, object, mixed, file, json

So oder so, danke für diesen Adapter :)

klein0r, 2020-10-27:
Keine Ahnung warum ich immer numeric statt number schreibe - irgendwie n Knoten im Kopf :) Danke

klein0r, 2020-10-29:
Gerade mal angeschaut, das ist doch in 0.0.6 schon gefixt?

https://github.com/klein0r/ioBroker.octoprint/commit/2816b5bcbd1375bab5a90ab730e160ec23c54884#

closed 2020-10-29 by klein0r

Jarvid, 2020-10-29:
Hallo,

kurios, diese Version habe ich auch installiert, aber trotzdem kommt die Warnung?
War auch kein altes Backup sondern ist eine neue ioBroker-Instanz ohne Backup-Einspielung.

Kann mir nur noch vorstellen, dass der Fehler bei ioBroker selbst liegt.
Dann entschuldige die Beunruhigung :)

klein0r, 2020-10-30:
Alles gut, lieber 2x schauen als den Fehler ewig mitzuschleppen 👍