klein0r/ioBroker.octoprint #24: Octoprint 1.0.3 Fehlermeldungen im Log

joergeli, 2020-12-06:
Moin,
nach dem Update auf Octoprint 1.0.3 bekomme ich **fortlaufend** folgende Fehlermeldungen im Log:
**`2020-12-06 05:41:16.141	error (7671) Cannot read property 'toFixed' of null `**

Das kommt anscheinend daher, weil noch keine Druckdatei in Octoprint zum Drucken ausgwählt wurde, also das Statusfeld leer ist:
[image]

Erst wenn ich **einmalig** eine Datei zum Drucken auswähle, d.h. eine Druckdatei im Statusfeld vorhanden ist, hören die Fehlermeldungen auf, also so:
[image]

Auch wenn ich anschl. die Druckdatei aus OctoPi lösche, kommen die Fehlermeldungen **nicht** wieder.
Es sieht so aus, als ob der Octoprint-Adapter beim Start zwingend eine aktivierte Druckdatei braucht?

Gruß
Jörg

h3ck0, 2020-12-07:
Moin

Das Problem bestand bei mir auch.

octoprint.0 | 2020-12-07 10:25:58.232 | error | (6858) Cannot read property 'toFixed' of null
octoprint.0 | 2020-12-07 10:25:57.887 | info | (6858) starting. Version 1.0.3 in /opt/iobroker/node_modules/iobroker.octoprint, node: v10.23.0, js-controller: 3.1.6

außerdem wurde mir in der Instance bei "Connected to octoprint" ein false angezeigt

nachdem ich eine priint datei rein geladen habe, war kam der Log Eintrag nicht mehr im Log und der adapter lief.

jedoch bin ich durch druck-file löschen nicht wieder in den Fehlerzustand gekommen.

klein0r, 2020-12-07:
Danke für die Info, scheinbar werden die Felder also nicht immer befüllt. Schaue ich mir an

FlexerJR, 2020-12-08:
Wollte auch gerade diesen Bug reporten, aber so kann ich es confirmen. Hier ein Screenshot von mir
Im Grunde läuft der Adapter jedoch kommen die Fehlermeldungen wenn ich einmal den Drucker iinkl Raspberry aus und wieder an mache.

Bin auf die Vorgängerversion zurück gegangen.
[image]

Smarte Grüße

klein0r, 2020-12-08:
Welche OctoPrint Version nutzt ihr? Bei 1.5.1 habe ich das Problem nicht / kann es nicht reproduzieren

closed 2020-12-08 by klein0r

h3ck0, 2020-12-08:
[image]

Ich nutze auch 1.5.1

klein0r, 2020-12-08:
Bitte mal die 1.0.4 testen ob der Fehler dann weg ist.

h3ck0, 2020-12-08:
Habe die 1.0.4 gerade erst gesehen.
Gucke ich mir morgen mal an...

joergeli, 2020-12-08:
Mit **OctoPrint V 1.4.2** (also noch nicht die aktuelle Version) und **Octoprint-Adapter 1.0.4** ist der Fehler noch vorhanden.
Ich werde mogen auf **OctoPrint V 1.5.1** updaten und berichten.

joergeli, 2020-12-09:
**@klein0r**
Ich habe jetzt auf **OctoPrint V 1.5.1** upgedatet.
**Octoprint-Adapter ist  1.0.4**
Leider ist die Fehlermeldung "`Cannot read property 'toFixed' of null`" immer noch vorhanden.
Diese Error-Meldung wird fortlaufend in's Log geschrieben, bis ich eine Druckdatei zum Druck auswähle, d.h. irgendeine Druckdatei im Statusfeld vorhanden ist ( siehe meinen zweiten Screenshot in meinem ersten Posting).
Anschließend hören die Error-Meldungen auf.

Komischerweise erscheinen die Fehlermeldungen **nicht** wieder, wenn ich die Druckdatei lösche.

____________________________________________________________________________

Einige Erläuterungen zu meiner Vorgehensweise, evtl. hilfreich bei der Fehlersuche? :
1.) WLAN-Steckdose wird via Script eingeschaltet und startet ihrerseits OctoPi-Rasp und Druckeri.
2.) Script wartet 30 Sekunden, bis OctoPi-Raspi **kpl. gebootet** ist, d.h. OctoPi Webseite erreichbar ist.
3.) Script startet Octoprint-Instanz.
4.) Error-Meldungen tauchen **fortwährend** im Log auf.
5.) Ich lade eine Druckdatei --> Errormeldungen hören auf.

Ich habe die Octopront-Instanz auch schon mal testweise vor dem Starten meines Scriptes aktiviert.
Dann tauchen halt die "üblichen" Log-Einträge
`

octoprint.0 | 2020-12-09 23:19:37.635 | info | (20476) timeout of 2000ms exceeded
-- | -- | -- | --
octoprint.0 | 2020-12-09 23:19:30.623 | info | (20476) timeout of 2000ms exceeded
octoprint.0 | 2020-12-09 23:19:23.619 | info | (20476) OctoPrint API not connected
octoprint.0 | 2020-12-09 23:19:23.615 | info | (20476) timeout of 2000ms exceeded

`
auf, weil der Adapter sich logischerweise nicht mit dem Octoprint-Raspi verbinden kann.
Wenn ich dann den Drucker, bzw, den Octoprint-Raspi einschaltet, tauchen die Errormeldungen wieder auf.
Deshalb kann es m.E. **nicht** am Startzeitpunkt der Octoprint-Instanz liegen ( hat ja auch bis Octoprint-Adapter 1.02  funktioniert ).
______________________________________________________________________
Du hast weiter oben geschrieben:
> 
> 
> Danke für die Info, scheinbar werden die Felder also nicht immer befüllt. Schaue ich mir an

ja, beim ersten Start von OctoPi ist noch nichts im Statusfeld vorhanden, oder ist das bei Dir anders? :
[image]

Ich als Laie würde vermuten, daß die Errormeldung von einem der leeren Felder im Screenshot verursacht wird.

Da fällt mir noch ein,  ab Adapter V 1.0.3 werden ja **Filament-Länge und Filament-Volumen** ausgelesen.
Evtl. kommt das Problem ja von diesen (noch) nicht vorhandenen Feldern nach dem Start?

Gruß
Jörg

klein0r, 2020-12-09:
> Da fällt mir noch ein, ab Adapter V 1.0.3 werden ja **Filament-Länge und Filament-Volumen** ausgelesen.
> Evtl. kommt das Problem ja von diesen (noch) nicht vorhandenen Feldern nach dem Start?

Richtig, daher habe ich in 1.0.4 ja eine Prüfung dafür eingebaut, ob die Felder wirklich vorhanden sind. Scheinbar sind diese ja da, nur eben leer. Kannst du mir einen Auszug aus dem Debug-Log geben? Am besten nur die eine Antwort von `/api/job`

joergeli, 2020-12-10:
Meinst Du das hier?:
`

octoprint.0 | 2020-12-10 01:04:13.648 | error | (15489) Cannot read property 'toFixed' of null
-- | -- | -- | --
octoprint.0 | 2020-12-10 01:04:13.637 | debug | (15489) received 200 response from /api/job with content:  {"job":{"estimatedPrintTime":null,"filament":{"length":null,"volume":null},"file":{"date":null,"name":null,"origin":null,"path":null,"size":n
octoprint.0 | 2020-12-10 01:04:13.561 | debug | (15489) received 200 response from /api/files?recursive=true with content: {"files":[],"free":13223653376,"total":15605321728}
__________________________________________________________________________
So sieht es übrigens aus, wenn eine Druckdatei aktiviert wurde, d.h. die Error-Medung nicht mehr auftritt.
Da ist zusätzlich noch **{"tool0":** drin.
Gilt wohl für Drucker mit nur einem HotEnd, es gibt ja auch Drucker mit mehreren HotEnds, wahrscheinlich wird dann durchnummeriert ( tool0, tool1, tool2, etc. )

octoprint.0 | 2020-12-10 01:21:09.930 | debug | (19198)  received 200 response from /api/job with content:  {"job":{"averagePrintTime":null,"estimatedPrintTime":447.4254598211647,"filament":{"tool0":{"length":161.6923217922449,"volume":0.388915610998
-- | -- | -- | --
__________________________________________________________________________

Ich weiß nicht, ob der obige Auszug ausreichend ist, deshalb sicherheitshalber noch ein längeres Debug-Log mit enthaltener Error-Meldung:

octoprint.0 | 2020-12-10 01:04:22.798 | error | (15489) Cannot read property 'toFixed' of null
-- | -- | -- | --
octoprint.0 | 2020-12-10 01:04:22.793 | debug | (15489) received 200 response from /api/job with content:  {"job":{"estimatedPrintTime":null,"filament":{"length":null,"volume":null},"file":{"date":null,"name":null,"origin":null,"path":null,"size":n
octoprint.0 | 2020-12-10 01:04:22.779 | debug | (15489) Registered system commands: core/shutdown, core/reboot, core/restart, core/restart_safe
octoprint.0 | 2020-12-10 01:04:22.778 | debug | (15489) received 200 response from /api/system/commands with content:  {"core":[{"action":"shutdown","confirm":"<strong>You are about to  shutdown the system.</strong></p><p>This action may  disrupt any
octoprint.0 | 2020-12-10 01:04:22.757 | debug | (15489) received 200 response from /api/printer with content:  {"sd":{"ready":true},"state":{"flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"pause
javascript.0 | 2020-12-10 01:04:22.745 | info | (851) script.js.common.PAW.OctoPi: Octoprint-Server ist online
octoprint.0 | 2020-12-10 01:04:22.738 | debug | (15489) received 200 response from /api/connection with content:  {"current":{"baudrate":250000,"port":"/dev/ttyUSB0","printerProfile":"_default","state":"Operational"},"options":{"baudratePreference"
octoprint.0 | 2020-12-10 01:04:22.589 | debug | (15489) re-creating refresh state timeout
octoprint.0 | 2020-12-10 01:04:22.588 | debug | (15489) sending get request to /api/job with data: null
octoprint.0 | 2020-12-10 01:04:22.587 | debug | (15489) sending get request to /api/system/commands with data: null
octoprint.0 | 2020-12-10 01:04:22.587 | debug | (15489) sending get request to /api/printer with data: null
octoprint.0 | 2020-12-10 01:04:22.586 | debug | (15489) sending get request to /api/connection with data: null
octoprint.0 | 2020-12-10 01:04:22.584 | debug | (15489) connected to OctoPrint API - online!
octoprint.0 | 2020-12-10 01:04:22.583 | debug | (15489) received 200 response from /api/version with content: {"api":"0.1","server":"1.5.1","text":"OctoPrint 1.5.1"}
octoprint.0 | 2020-12-10 01:04:22.549 | debug | (15489) sending get request to /api/version with data: null
octoprint.0 | 2020-12-10 01:04:22.547 | debug | (15489) refreshing OctoPrint state
octoprint.0 | 2020-12-10 01:04:17.794 | error | (15489) Cannot read property 'toFixed' of null
octoprint.0 | 2020-12-10 01:04:17.791 | debug | (15489) received 200 response from /api/job with content:  {"job":{"estimatedPrintTime":null,"filament":{"length":null,"volume":null},"file":{"date":null,"name":null,"origin":null,"path":null,"size":n
octoprint.0 | 2020-12-10 01:04:17.774 | debug | (15489) Registered system commands: core/shutdown, core/reboot, core/restart, core/restart_safe
octoprint.0 | 2020-12-10 01:04:17.774 | debug | (15489) received 200 response from /api/system/commands with content:  {"core":[{"action":"shutdown","confirm":"<strong>You are about to  shutdown the system.</strong></p><p>This action may  disrupt any
octoprint.0 | 2020-12-10 01:04:17.753 | debug | (15489) received 200 response from /api/printer with content:  {"sd":{"ready":true},"state":{"flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"pause
javascript.0 | 2020-12-10 01:04:17.694 | info | (851) script.js.common.PAW.OctoPi: Octoprint-Server ist online
octoprint.0 | 2020-12-10 01:04:17.689 | debug | (15489) received 200 response from /api/connection with content:  {"current":{"baudrate":250000,"port":"/dev/ttyUSB0","printerProfile":"_default","state":"Operational"},"options":{"baudratePreference"
octoprint.0 | 2020-12-10 01:04:17.546 | debug | (15489) re-creating refresh state timeout
octoprint.0 | 2020-12-10 01:04:17.545 | debug | (15489) sending get request to /api/job with data: null
octoprint.0 | 2020-12-10 01:04:17.544 | debug | (15489) sending get request to /api/system/commands with data: null
[... cut ...]
octoprint.0 | 2020-12-10 01:04:12.506 | debug | (15489) received 200 response from /api/version with content: {"api":"0.1","server":"1.5.1","text":"OctoPrint 1.5.1"}
octoprint.0 | 2020-12-10 01:04:12.233 | debug | (15489) sending get request to /api/version with data: null
octoprint.0 | 2020-12-10 01:04:12.222 | debug | (15489) refreshing OctoPrint state
octoprint.0 | 2020-12-10 01:04:12.220 | debug | (15489) Starting with API refresh interval: 5 (5 while printing)
octoprint.0 | 2020-12-10 01:04:12.191 | info | (15489) starting. Version 1.0.4 in /opt/iobroker/node_modules/iobroker.octoprint, node: v12.16.3, js-controller: 3.1.6

`

klein0r, 2020-12-10:
Ahhh okay, danke! Da wird also null übertragen. Ich dachte, dann gibt es die Werte gar nicht mehr. Mache mal gerade 1.0.5 fertig 😄 

```"filament":{"length":null,"volume":null}```

klein0r, 2020-12-10:
So, bitte 1.0.5 testen

joergeli, 2020-12-10:
Ist Version 1.0.5 schon im latest Repository (http://download.iobroker.net/sources-dist-latest.json) verfügbar, oder dauert das eine gewisse Zeit?

FlexerJR, 2020-12-10:
1.0.5 Sieht gut aus! Danke!

joergeli, 2020-12-10:
@FlexerJR
> 
> 
> 1.0.5 Sieht gut aus! Danke!

Wo hast Du die her? Direkt installiert?
Bei mir ist 1.05 noch nicht im ioBroker latest Repository vorhanden.

FlexerJR, 2020-12-10:
Hier den Link kopieren und über die Katze in iob installieren

klein0r, 2020-12-10:
> Hier den Link kopieren und über die Katze in iob installieren

Bitte nicht! Einfach einen Tag warten oder auf Refresh in der Adapter-Liste klicken. Im latest-"Verwahrungsort" ist die Version dann sichtbar. Über die Katze installiert man ausschließlich als ENTWICKLER Adapter. Niemals als Anwender. Damit holst Du Dir einen undefinierten Stand aus GitHub und keine spezifische Version.

FlexerJR, 2020-12-10:
Ja korrekt!

joergeli, 2020-12-10:
> 
@FlexerJR 

> 
> Hier den Link kopieren und über die Katze in iob installieren

Das wollte ich vermeiden.

@klein0r 
Refresh in der Adapter-Liste zeigt noch nichts.
OK, dann warte ich ab.

joergeli, 2020-12-10:
@klein0r 
Alles paletti, mit V 1.0.5 ist der Fehler behoben.

**Vielen Dank für Deine Mühe!**

Gruß
Jörg

baeriwiliwonka, 2020-12-17:
Nach dem Update auf 1.0.5 ist der Fehler "Cannot read property 'toFixed' of null" verschwunden.
Mein Adapter ist aber immer noch gelb, habe jetzt folgende Meldungen: 

`

octoprint.0 | 2020-12-17 20:35:24.907 | info | (23965) OctoPrint API not connected
-- | -- | -- | --
octoprint.0 | 2020-12-17 20:35:24.904 | info | (23965) timeout of 2000ms exceeded

`

commit 7709e6a9 (2020-12-08): Fixed .toFixed exception when no job is running - fixes #24