klein0r/ioBroker.octoprint #27: Adapter bleibt Gelb

staysafe-de, 2020-12-16:
Hallo ich habe ein kleines Problem:
Adapter bleibt gelb.
Adapter Version 1.0.3
Admin 4.1.10
octoprint Version 1.5.2

octoprint.0 | 2020-12-16 11:25:29.365 | error | (29690) Cannot read property 'toFixed' of null
-- | -- | -- | --
octoprint.0 | 2020-12-16 11:24:29.419 | error | (29690) Cannot read property 'toFixed' of null
octoprint.0 | 2020-12-16 11:24:28.941 | info | (29690) starting. Version 1.0.3 in /opt/iobroker/node_modules/iobroker.octoprint, node: v10.20.1, js-controller: 3.1.6
octoprint.0 | 2020-12-16 10:24:18.536 | info | (29674) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
octoprint.0 | 2020-12-16 10:24:18.534 | info | (29674) terminating
octoprint.0 | 2020-12-16 10:24:18.532 | info | (29674) Got terminate signal TERMINATE_YOURSELF

staysafe-de, 2020-12-16:
jetzt geht es auf einmal - mehrere Stunden Später

closed 2020-12-16 by staysafe-de

klein0r, 2020-12-16:
Den Fehler bekomme ich auch manchmal noch. Muss ich mir anschauen woran das liegt

Totalausfall, 2021-01-01:
Frohes Neues :)

Gibt es hierzu bereits eine Idee? Ich habe leider das selbe Problem.