klein0r/ioBroker.octoprint #34: Pfad zur aktuellen Druckdatei

beef76, 2021-01-19:
Hi Matthias
wäre es möglich den kompletten Pfad der aktuell geladenen bzw. druckenden Datei auszulesen und in einem DP zur Verfügung zu stellen? Aktuell wird ja nur der Filename ausgelesen, egal wo der gcode in Octoprint liegt. 
Für die Weiterverarbeitung in der VIS wäre es hilfreich den Pfad zum Gcode zu haben, wenn dieser in einem Unterordner im Octoprint liegt.
VG Björn

klein0r, 2021-01-19:
Siehe https://docs.octoprint.org/en/master/api/job.html#retrieve-information-about-the-current-job

Ich nehme (glaube ich) 1:1 den Namen. Mehr gibt es nicht.

beef76, 2021-01-19:
hab nen Weg gefunden mir die aktuellen Druckinfos per http get zu ziehen und da den Pfad zum file raus zu ziehen. ich schließe dann mal :-)

closed 2021-01-19 by beef76