klein0r/ioBroker.octoprint #37: GPIO auslesen

fulwerin, 2021-02-06:
Guten Tag

Könnte man irgendwie die GPIO Pins vom Octopi Server auf dem ioBroker Server auslesen?
Wenn das nicht geht könnte man das im Adapter hinzufügen?

vielen Dank

klein0r, 2021-02-06:
Der Adapter bedient nur die offizielle API von OctoPrint. Da gibt es keine Möglichkeit die GPIO Pins auszulesen.
Außer es gibt ein Plugin für OctoPrint, welches die API erweitert.

closed 2021-02-06 by klein0r

mczeus, 2021-02-14:
Hätte gerne etwas ähnliches!
Habe das OctoPrint-Enclosure Plugin am laufen mit dem ich Lüfter und Beleuchtung steuern kann. Es gibt dafür auch eine API, ist es möglich das in den Adapter zu integrieren?
https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/API-Control