klein0r/ioBroker.octoprint #44: octoprint.1 als 2te octoprint Instanz geht im ioBroker nicht

dg1bor, 2021-05-03:
Hallo Matthias,

meine erste octoprint.0 Instance mit einem Creality Ender 3Plus funktioniert ohne Probleme,
jedoch wollte ich eine zweite Instance octoprint.1 für einen Prusa MK3S dazu packen.
Die API wird jedoch nicht erkannt.

Beim Ender3 (Merlin Firmware) läuft ein Raspi mit OctoPrint 1.6.0 
Beim Prusa (Prusa Firmware) läuft ein Raspi mit OctoPrint 1.6.0

Deine  octoprint wird immer aktuell gehalten.

Die API wurde von Octoprint direkt in die Instanz kopiert, eine Verbindung wird aber nicht hergestellt. gelber Punkt
API neu erzeugt und kopiert, geht auch nicht 3 mal versucht.
In der ioBroker Instanz sind bei beiden bis auf die IP und API die gleichen Werte enthalten.

wo kann der Fehler liegen

Gruß
Gerhard Bauer

[image]

klein0r, 2021-05-03:
Ich habe auch Probleme seit dem Update auf 1.6.0 gestern. Muss ich mir angucken

klein0r, 2021-05-03:
Bitte mit 1.1.0 testen

closed 2021-05-03 by klein0r