klein0r/ioBroker.octoprint #45: API-Key geht verloren.

Brainbug01, 2021-05-04:
octoprint.0 | 2021-05-04 19:29:21.597 | error | (22753) Invalid value "undefined" for header "X-Api-Key"
-- | -- | -- | --

Ich gebe den API Key und Instanz wird grün, später schaue ich wieder nach und dann ist er gelb und die Meldung im Log.

klein0r, 2021-05-04:
Du hast den also nach dem Update neu konfiguriert?

Brainbug01, 2021-05-04:
Ja und wenn ich die Instant zweimal neu starte bleibt sie gelb mit dem Fehler im Log.

Brainbug01, 2021-05-04:
Und hiermit wird es ebenfalls gelöscht:

```
var obj = getObject("system.adapter.octoprint.0");
obj.common.enabled = true; // Adapter einschalten
setObject("system.adapter.octoprint.0", obj);
```

klein0r, 2021-05-04:
Was sind das denn für Scripts? Der Key ist jetzt als `protectedNative` konfiguriert. Das heißt, dass andere Adapter den Key aus Sicherheitsgründen nicht lesen dürfen. In Deiner Antwort von getObject ist also der Key schon nicht mehr enthalten. Wenn Du dann das Objekt neu schreibst, dann ist er logischerweise weg.

Brainbug01, 2021-05-04:
Das Script findest du in er Beschreibung:
https://youtu.be/g-MaFbpXNqI

klein0r, 2021-05-04:
@GermanBluefox Ist das der offizielle Weg eine Instanz aus einem Script heraus zu stoppen? Zusammen mit protectedNative wird das ja an vielen Stellen zu Problemen führen. bzw. aus allen Adaptern heraus, die nicht Admin sind.

klein0r, 2021-05-05:
Bitte einfach `true|false` an `system.adapter.octoprint.0.alive` senden und den anderen Code für keinen Adapter mehr verwenden 😄

closed 2021-05-05 by klein0r

StrayDog2205, 2021-05-05:
Moin
Das war (zumindest damals) der Weg den Adapter zu stoppen.

Das Skript ist ja aber mittlerweile auch schon über zwei Jahre alt.
Da hat sich ja einiges getan.

Ich werde das Skript anpassen und neu hochladen.

klein0r, 2021-05-05:
Danke Sven! Perfekt

foxriver76, 2021-05-14:
> Das CLI macht es ja scheinbar genauso:
> https://github.com/ioBroker/ioBroker.js-controller/blob/5884e470359a08df6c4f5fba8b7bc93457434baf/lib/cli/cliProcess.js#L316

Controller arbeitet intern mit der DB direkt und nicht mit Adapter Methoden, daher klappt das im Controller - Controller Code sollte nicht zwingend als Beispiel dienen.

Die einfachste Möglichkeit wäre wohl `extendObject('system.adapter.octoprint.0', {common: {enabled: false}})`.

klein0r, 2021-05-14:
Wird bei extend also die protected Property nicht „gelöscht“? Also passiert das extend auf Controller Seite und nicht einfach nur eine Kurzform für lesen, ändern, schreiben im Adapter Code?

foxriver76, 2021-05-14:
> Sieht ja erstmal so aus
> 
> https://github.com/ioBroker/ioBroker.js-controller/blob/5505a2233f8511c6ffa7506a1846a0739a1ca9d7/lib/adapter.js#L2164

Genau, passt doch. `adapterObjects` ist eine Instanz der Objects DB, es wird ja nicht `this.getForeignObjectAsync` genutzt.

klein0r, 2021-05-14:
> Genau, passt doch.

Ne, wird nicht klappen. In dem getObject sind die protected native Attribute ja schon nicht enthalten. Führt also zum gleichen Fehlerbild wie der Code weiter oben.

foxriver76, 2021-05-14:
> Ne, wird nicht klappen.

Gut zu wissen - Danke!

Apollon77, 2021-05-14:
> Die einfachste Möglichkeit wäre wohl extendObject('system.adapter.octoprint.0', {common: {enabled: false}}). Edit: Oder .alive falls Ingo das gesagt hat ;-)

Ingo hat beide Optionen genannt :-))

klein0r, 2021-05-18:
@foxriver76 

> Gut zu wissen - Danke!

Entschuldige, sollte nicht 9-mal-klug klingen. Du bist ja viel tiefer im Thema als ich. Habe es gerade doch noch mal getestet. Aber bei:

```javascript
this.extendObject('system.adapter.lametric.0', {common: {enabled: false}});
```

*(eine Instanz stoppt die andere - wie beim Ausgangsproblem die JavaScript-Instanz die Octoprint-Instanz)*

Bekomme ich den Fehler: `(1090) Object octoprint.0.system.adapter.lametric.0 is invalid: obj.type has to exist`

Logisch, es wird ja nicht foreign Object genutzt (wie Du sagtest), dadurch wird der Namespace vorangestellt. Hätte mich auch gewundert wenn das klappt.

foxriver76, 2021-05-18:
Geht ja beim Fragesteller um den javascript Adapter, da ist ein `extendObject` quasi ein èxtendForeignObject`, daher geht es dort. Im Adapterfall versucht jetzt der lametric Adapter ein `lametric.0.syste.adapter.lametric.0`Objekt zu extenden (wie du schon sagtest), da es wohl nicht existiert, wird probiert das Objekt zu erstellen, und das tut er nicht, da er dann ein Objekt erstellen würde, welches nur `common.enabled`als Attribut hätte, somit invalide ist, somit alles as-designed. 

Wenn du testen möchtest aus dem Adapter heraus bitte `extendForeignObject` nutzen. hm-rpc tut das schon lange und es klappt, siehe https://github.com/ioBroker/ioBroker.hm-rpc/blob/15ef687e09b8d6bd5cb4ce654d1b535cb4a1c52e/main.js#L1376

> Entschuldige, sollte nicht 9-mal-klug klingen.

Schon gut, nur da ich 100 % weiß dass es geht, konnte ich mir den Sarkasmus nicht ganz verkneifen ;-)