klein0r/ioBroker.octoprint #50: Adapter bleibt gelb bei Octoprint Neustart

webwalker2001, 2021-05-17:
Hi Matthias,

ich benutze den Octoprint Adapter schon länger, aber seit kurzem wird der Adapter nicht mehr grün wenn ich den Raspbery-Pi mit Octoprint starte.
Starte ich dann den Adapter neu wird er grün.
Fahre ich dann den Rasperry-Pi mit Octoprint runter und starte neu bleibt der Adapter gelb.

Mein System:
Raspberry-Pi 4/4GB
Octoprint-Adapter V1.1.0
Node.js V12.22.1
NPM V6.14.12
js-controller V3.3.10
OS: Raspbian GNU/Linux
Kernel 5.10.17-v7l+

Octoprint: V1.6.1

LG

webwalker2001

klein0r, 2021-05-18:
Du stoppst den Adapter aber nicht zufällig mit einem Script?

klein0r, 2021-05-18:
Schau mal bitte #45 an

webwalker2001, 2021-05-18:
Hi, ich stoppe den Adapter eigentlich gar nicht. Wenn er gelb ist starte ich ihn von Hand neu. 
Den Raspberry-Pi mit Octoprint schalte ich nur ein wenn ich den Drucker brauche.
Hat vorher einwandfrei geklappt. Weiß aber nicht genau seit wann nicht mehr (update Octoprint oder update Octoprint-Adapter)
Der API-Key muß nicht neu eingetragen werden.

klein0r, 2021-05-18:
Kannst Du mal das Log Level auf Debug stellen für die Instanz und den Inhalt teilen?

webwalker2001, 2021-05-18:
Bitte schön

[Log.txt](https://github.com/klein0r/ioBroker.octoprint/files/6502682/Log.txt)

Ceel84, 2021-05-21:
Ich habe das gleiche Problem
Nur passiert es nicht nur wenn ich Octoprint neu starte, sondern bei mir wird der Adapter einfach zwischendurch gelb.
Ohne ein Script oder sonst eine Änderung.

Nutze Admin 5.x

Z18Aplha, 2021-05-21:
Ich hab genau das gleiche Problem wie @Ceel84 

Admin 5.x und sowohl beim global api key, als auch bei einem anwendungsspezifischen api key wird der Adapter nach einiger Zeit wieder gelb. Ein manueller restart lässt ihn wieder grün werden

HiroP77, 2021-05-21:
Ich auch, gefühlt nach dem Update von Octoprint auf 1.6.1
Adapter wird erst grün, nach etwa 10 min wird er gelb und wird erst wieder nach Neustart grün.
logs sagen nur, dass die API nicht mehr connected wäre.

klein0r, 2021-05-27:
Bitte 1.1.1 testen

closed 2021-05-27 by klein0r