klein0r/ioBroker.octoprint #51: API Schlüssel verschwindet und Instanz bleibt gelb

Ceel84, 2021-05-18:
In Admin 5.0.27 verschwindet bei mir der API Schlüssel immer, wodurch iobroker immer die Verbindung verliert.

 - Adapter version: 1.1.0
 - JS-Controller version: 3.3.10
 - Node version: 12.22.1
 - Operating system: Proxmox vm

klein0r, 2021-05-18:
Lass mich raten: Du stoppst und startest den Adapter mit einem Script?

Ceel84, 2021-05-18:
Hallo
Ja so sollte es laufen.
Habe das Script aber deaktiviert und habe den gleichen Fauler.

klein0r, 2021-05-18:
Du hast das Script deaktiviert, dann den API-Key hinterlegt und gespeichert?
Wann genau "verschwindet" er dann wieder?

Ceel84, 2021-05-18:
Ja habe ich.
Kann ich nicht genau sagen.
Irgendwann ist er wieder gelb.

Ceel84, 2021-05-18:
Warum sollte das Script ein Problem verursachen?

klein0r, 2021-05-18:
see #45

Ceel84, 2021-05-18:
Das werde ich Mal testen danke.
Kann sein das es nur daran liegt
Sorry dann

klein0r, 2021-05-18:
Ich dachte das hast Du gerade ausgeschlossen?!

Ceel84, 2021-05-18:
Du hast recht!
 Es liegt wirklich am Script.
habe einfach Adaper deaktiviert über den alive dp

danke und sorry nochmal

bin bei dem ganzen hin und her Probieren durcheinander gekommen was ich genau getestet hatte und was nicht!

Ceel84, 2021-05-18:
Es besteht trotz Deaktivierung des Scripts das die Instanz gelb wird.
Der API Code bleibt aber bestehen.

bluefish007, 2021-05-18:
Habe leider das selbe Problem! Instanz wird während dem Drucken irgendwann gelb... (seit Admin 5 Version)

klein0r, 2021-05-19:
duplicate of #50

closed 2021-05-19 by klein0r