klein0r/ioBroker.octoprint #56: Long file names not supported?

beef76, 2021-09-16:
### The problem

Is there any known file/path length limitation while starting prints from the instance Datapoints? Or any characters are maybe not supported?

I'm usually exporting Gcodes via SuperSlicer including variables for Material, Printtime ect. (all seperated with an underscore, no special characters beside the underscore, no space in between ect.) and found out, that starting the print via the datapoint (scheduled prints mostly) is not working with such long filenames.
Using a short filename like "Test.gcode" or similar it's working perfectly fine.

Is this a known "issue" already? Any workarounds available? Don't want to miss all the information in the filenames and especially don't want to rename all stored gcodes on the different octoprint instances i have.

Thanks in advance!
Bjoern

### Version of ioBroker js-controller

3.3.18

### Version of Adapter

1.1.1

### Operating system running ioBroker

Linux

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

there is certainly nothing reported in the logs when trying to print a file with long filename so nothing to provide her unfortunately

beef76, 2021-09-16:
just to add, i want to create automation function to run printjobs on my printfarm with schedules, without the possibility to run also files with long filenames a lot of rework to rename all gcodes would be needed

klein0r, 2021-09-18:
Please test 1.1.2

beef76, 2021-09-20:
1.1.2 seems to solve this issue as well, thanks!

closed 2021-09-20 by beef76