klein0r/ioBroker.octoprint #58: [Request] Daten aus Plugin DisplayLayerProgress

neo862, 2021-09-23:
Besteht die Möglichkeit die Daten aus dem Octoprint Plugin (https://github.com/OllisGit/OctoPrint-DisplayLayerProgress) auszulesen und in den IOBroker zu bekommen.

Das würde z.B. Layer oder Höhenabhängige Smarthomebefehle erlauben. (z.B. Im firstLayer Licht 100% anschließend auf 50% dimmen, etc.)

Gruß
Stefan

klein0r, 2021-12-12:
Kommt in die nächste Version, muss es noch testen

joergeli, 2021-12-27:
Ich habe die aktuelle Version des Octoprint-Adapters **V 2.0.7** installiert,
habe aber Probleme mit dem DisplayLayerProgress-Plugin.

Ich habe das Problem im ioBroker-Forum unter:
https://forum.iobroker.net/topic/50701/octoprint-2-0-7-problem-mit-display-layer-progress
näher beschrieben (geht für mich dort enfacher, als hier)

Negalein, 2021-12-27:
gleiches Problem mit 2.0.7 wie @joergeli 

Gesamtlayer passt. Aber der aktuelle Layer wird mit `-1` angezeigt.

[image]
[image]

klein0r, 2021-12-27:
Habe den Fehler schon behoben, Fix kommt in die nächste Version

joergeli, 2021-12-27:
Danke dafür !

klein0r, 2021-12-28:
Bitte 2.1.0 testen

closed 2021-12-28 by klein0r

joergeli, 2021-12-29:
Mit V2.1.0 funktioniert die Layer-Anzeige korrekt .