klein0r/ioBroker.octoprint #61: Webcam-Stream in Vis

fetzer91, 2021-12-05:
### The problem

Hallo Matthias,
ich habe keine Probleme mit deinem Adapter und habe ihn soweit es für mich nötig war, in meinen Iobroker integriert und auch einen View in der Vis ausschließjich dem Octoprint gewidmet. Nun wäre das Tüpfelchen auf dem i noch auch den Webstream anzeigen zu lassen. Hast du dazu eine Idee bzw. könntest du das noch mit integrieren?
irgendwie muss es möglich sein, denn verschiedene Apps für Handys können den Stream auch anzeigen.

Freundliche Grüße und vielen Dank für deine bislang geleistete Arbeit
Stefan

### Version of ioBroker js-controller

v14.18.2

### Version of Adapter

1.1.1

### Operating system running ioBroker

Linux Ubuntu 20.04 LTS iobroker unter Proxmox 6.3-6

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2021-12-06:
Was genau wünschst Du Dir denn als Datenpunkt? Nur die URL zum Stream?

fetzer91, 2021-12-06:
Hallo Matthias, die Url zum Stream auf jeden Fall. Und ein Widget welches den Stream dann auch in der vis anzeigen kann. Vielleicht gibt es das schon, aber dann habe ich das bislang nicht gefunden.
Danke und Gruß Stefan

klein0r, 2021-12-06:
> Hallo Matthias, die Url zum Stream auf jeden Fall. Und ein Widget welches den Stream dann auch in der vis anzeigen kann. Vielleicht gibt es das schon, aber dann habe ich das bislang nicht gefunden.

Warum nicht einfach per iFrame?

fetzer91, 2021-12-06:
Das habe ich versucht, aber iFrame kann mit der Webcam- Url von octoprint nichts anfangen.

klein0r, 2021-12-06:
Was gibt die denn für ein Format aus? Das muss gehen 😄

fetzer91, 2021-12-06:
Das weiß ich nicht. Ich nutze eine übliche Webcam, welche an Raspi via USB angeschlossen ist. Was der Raspi daraus macht und was dann an den Browser gesendet wird weiß ich nicht.

klein0r, 2021-12-07:
Naja, wenn Du die Webcam URL im Browser aufrufen kannst (und dann den Stream siehst), dann klappt das auch im iFrame.
Außer deine Visualisierung wird über HTTPS aufgerufen und der Stream im iFrame soll per HTTP aufgerufen werden - das erlauben die Browser Policies nicht.

fetzer91, 2021-12-07:
Hallo Matthias, genau das geht aber nicht. Die Webcam hat keine gewöhnliche IP ala 192.168….. sondern „ /webcam/?action=stream“. Und das lässt sich nicht einfach über nen Browser aufrufen.
Wenn es so einfach wäre, dann würde ich dich damit nicht behelligen.
Freundliche Grüße Stefan

klein0r, 2021-12-07:
> sondern „/webcam/?action=stream“

Das klingt doch erstmal wie eine ganz normale URL vom mJPEGStreamer? Warum sollte man die nicht im Browser aufrufen können? Habe ich schon ganz oft gemacht und sogar auf YouTube gezeigt.

> Wenn es so einfach wäre, dann würde ich dich damit nicht behelligen.

Das geht wirklich so einfach - was spricht aus Deiner Sicht dagegen? Ich konfiguriere den immer auf Port 8080 also zum Beispiel:

``http://<ipvonoctoprint>:8080/webcam/?action=stream``

Und die URL in einem iFrame angeben. Fertig.

Siehe auch https://youtu.be/rIUK0h9oKj4?t=1191

fetzer91, 2021-12-07:
Hmm…ich glaube ich hatte ein Brett vorm Kopf. Ich werde mir morgen nochmal das Video zu Gemüte führen und auch gleich mal ein neues Octoprint aufsetzen. Meine aktuelle Version ist schon mächtig alt.
Wenn es dann wirklich so einfach geht, dann braucht es den extra Datenpunkt wahrscheinlich doch nicht.
Danke dir, ich berichte morgen ob es funktioniert hat.
Grüße Stefan

closed 2021-12-12 by klein0r