klein0r/ioBroker.octoprint #63: timeout of 2000ms exceededBug report - only 2.0.6

OberfragGER, 2021-12-20:
### The problem

Hi,
ich bekomme aktuell immer timeouts beim verbinden. Habe bereits den Adater neu installiert.

browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/96.0.4664.110 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 900.0
env.hardware.ram: 915730432
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 2 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: True
octoprint.safe_mode: False
octoprint.version: 1.7.2
systeminfo.generator: zipapi

### Version of nodejs

v14.18.2

### Version of ioBroker js-controller

3.3.22

### Version of Adapter

2.0.6

### Operating system running ioBroker

Raspian Buster

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

[iobroker.2021-12-20.log.txt](https://github.com/klein0r/ioBroker.octoprint/files/7748253/iobroker.2021-12-20.log.txt)

OberfragGER, 2021-12-28:
Its still in 2.07. , 2.10.
In the meantime i tried 2.0.4
-Its also there. So title is misleading.

Ralf-Niemeier, 2022-01-14:
Sometimes ! same here.
I can't find no reason or solution 
Adapter Version 2.1.0
thx.

klein0r, 2022-01-14:
Please test with 2.1.1

If timeout still exceeded:

- Increase timeout in instance config
- Provide debug log

closed 2022-01-14 by klein0r

OberfragGER, 2022-01-17:
Was not working.
I updatet to version 2.20

<html>
<body>
<div class="MuiGrid-root-335 MuiGrid-container-336 MuiGrid-item-337 MuiGrid-align-items-xs-center-344" style="box-sizing: border-box; width: 1724px; display: flex; flex-wrap: wrap; margin: 0px; align-items: center; color: rgba(0, 0, 0, 0.87); font-family: &quot;Segoe UI&quot;, Tahoma, Arial, &quot;Courier New&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><span class="MuiBadge-root-106" title="Fehler anzeigen" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;"><span class="MuiBadge-badge-107 jss5329 jss5330 MuiBadge-anchorOriginTopRightRectangle-112 MuiBadge-colorError-110 MuiBadge-invisible-128" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 195ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; color: rgb(255, 255, 255); background-color: rgb(244, 67, 54); top: 10px; right: 10px; transform: scale(0) translate(50%, -50%); transform-origin: 100% 0%;">0</span></span><span class="MuiBadge-root-106" title="Fehler und Warnungen anzeigen" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;"><button class="MuiButtonBase-root-85 MuiIconButton-root-77" tabindex="0" type="button" style="box-sizing: inherit; color: rgba(0, 0, 0, 0.54); border: 0px; cursor: pointer; margin: 0px; display: inline-flex; outline: 0px; padding: 12px; position: relative; align-items: center; user-select: none; border-radius: 50%; vertical-align: middle; justify-content: center; text-decoration: none; background-color: transparent; appearance: none; -webkit-tap-highlight-color: transparent; flex: 0 0 auto; overflow: visible; font-size: 1.5rem; text-align: center; transition: background-color 150ms cubic-bezier(0.4, 0, 0.2, 1) 0ms;"><span class="MuiIconButton-label-84" style="box-sizing: inherit; width: 24px; display: flex; align-items: inherit; justify-content: inherit;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M1 21h22L12 2 1 21zm12-3h-2v-2h2v2zm0-4h-2v-4h2v4z"></path></svg></span><span class="MuiTouchRipple-root-181" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></button><span class="MuiBadge-badge-107 jss5329 jss5331 MuiBadge-anchorOriginTopRightRectangle-112" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 225ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; top: 10px; right: 10px; transform: scale(1) translate(50%, -50%); transform-origin: 100% 0%; background-color: rgb(255, 193, 7);">4</span></span><div class="jss5318" style="box-sizing: inherit; flex-grow: 1;"></div><div style="box-sizing: inherit;"><button class="MuiButtonBase-root-85 MuiButton-root-136 MuiButton-contained-144 MuiButton-containedPrimary-145" tabindex="0" type="button" style="box-sizing: border-box; color: rgb(255, 255, 255); border: 0px; cursor: pointer; margin: 0px; display: inline-flex; outline: 0px; padding: 6px 16px; position: relative; align-items: center; user-select: none; border-radius: 4px; vertical-align: middle; justify-content: center; text-decoration: none; background-color: rgb(51, 153, 204); appearance: none; -webkit-tap-highlight-color: transparent; font-size: 0.875rem; min-width: 64px; transition: background-color 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms, box-shadow 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms, border 250ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1.75; letter-spacing: 0.02857em; text-transform: uppercase; box-shadow: rgba(0, 0, 0, 0.2) 0px 3px 1px -2px, rgba(0, 0, 0, 0.14) 0px 2px 2px 0px, rgba(0, 0, 0, 0.12) 0px 1px 5px 0px;"><span class="MuiButton-label-137" style="box-sizing: inherit; width: 188.719px; display: inherit; align-items: inherit; justify-content: inherit;"><span class="MuiButton-startIcon-158 MuiButton-iconSizeMedium-161" style="box-sizing: inherit; display: inherit; margin-left: -4px; margin-right: 8px;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M19 12v7H5v-7H3v7c0 1.1.9 2 2 2h14c1.1 0 2-.9 2-2v-7h-2zm-6 .67l2.59-2.58L17 11.5l-5 5-5-5 1.41-1.41L11 12.67V3h2z"></path></svg></span>LOG HERUNTERLADEN</span><span class="MuiTouchRipple-root-181" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></button></div><div class="jss5318" style [... cut ...]
octoprint.0 | 2022-01-17 17:05:12.656 | debug | Objects connected to redis: 127.0.0.1:9001
octoprint.0 | 2022-01-17 17:05:12.622 | debug | Objects client initialize lua scripts
octoprint.0 | 2022-01-17 17:05:12.621 | debug | Objects create User PubSub Client
octoprint.0 | 2022-01-17 17:05:12.620 | debug | Objects create System PubSub Client
octoprint.0 | 2022-01-17 17:05:12.618 | debug | Objects client ready ... initialize now
octoprint.0 | 2022-01-17 17:05:12.576 | debug | Redis Objects: Use Redis connection: 127.0.0.1:9001

</div></div>
</body>
</html>

klein0r, 2022-01-17:
@OberfragGER Ist dein System stark ausgelastet? 2 Sekunden ist ja eigentlich schon sehr lange für den Request. Sonst stell den Timeout doch einfach mal höher in der Config. Zumal es auch immer andere Ressourcen sind die in Timeouts laufen.