klein0r/ioBroker.octoprint #65: Wunsch: Slicer Thumbnails

Negalein, 2022-01-13:
### The problem

Hallo

Ist es möglich, dieses Plugin in den Adapter zu integrieren?

LG

### Version of nodejs

1

### Version of ioBroker js-controller

2

### Version of Adapter

3

### Operating system running ioBroker

4

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2022-01-15:
Please provide a gcode file with embedded image

Negalein, 2022-01-15:
> Please provide a gcode file with embedded image

https://www.nega.at/Monarch.gcode

klein0r, 2022-01-15:
Thanks - I will check if the plugin extends the API

klein0r, 2022-01-15:
Example OctoPrint image path:

http://172.16.0.160:5000/plugin/prusaslicerthumbnails/thumbnail/Monarch.png

klein0r, 2022-01-15:
Files array is extended:

```
{
            "DisplayLayerProgress": {
                "totalLayerCountWithoutOffset": "74"
            },
            "date": 1642252492,
            "display": "Monarch.gcode",
            "gcodeAnalysis": {
                "dimensions": {
                    "depth": 179.826,
                    "height": 8.96,
                    "width": 188.759
                },
                "estimatedPrintTime": 35686.73681764925,
                "filament": {
                    "tool0": {
                        "length": 22130.25471998654,
                        "volume": 53.22950057607261
                    }
                },
                "printingArea": {
                    "maxX": 190.759,
                    "maxY": 189.826,
                    "maxZ": 8.96,
                    "minX": 2,
                    "minY": 10,
                    "minZ": 0
                }
            },
            "hash": "da8f6db1870f882f10b15450002837d1745b4602",
            "name": "Monarch.gcode",
            "origin": "local",
            "path": "Monarch.gcode",
            "refs": {
                "download": "http://172.16.0.160:5000/downloads/files/local/Monarch.gcode",
                "resource": "http://172.16.0.160:5000/api/files/local/Monarch.gcode"
            },
            "size": 23092227,
            "thumbnail": "plugin/prusaslicerthumbnails/thumbnail/Monarch.png?20220115141452",
            "thumbnail_src": "prusaslicerthumbnails",
            "type": "machinecode",
            "typePath": [
                "machinecode",
                "gcode"
            ]
        }
```

See ``thumbnail`` and ``thumbnail_src``

closed 2022-01-15 by klein0r

klein0r, 2022-01-15:
@Negalein  Was genau möchtest Du denn haben? Einfach nur die URL zum Bild in einem Datenpunkt?

Negalein, 2022-01-15:
> Was genau möchtest Du denn haben? Einfach nur die URL zum Bild in einem Datenpunkt?

Ja, das würde reichen

klein0r, 2022-01-15:
Siehe 2.2.0

Negalein, 2022-01-15:
> Siehe 2.2.0

in welchemmDP ist der Bildlink zu finden?
[image]
[image]

klein0r, 2022-01-15:
Direkt unter dem betreffenden file

Negalein, 2022-01-15:
> Direkt unter dem betreffenden file

Ah, gefunden.
plugin/prusaslicerthumbnails/thumbnail/Kinder/Monarch.png?20220114081918

Das ist dann eher schlecht platziert.

Perfekt wäre unter `octoprint.0.printjob.file`
Und komplett inkl. IP vom Octo `http://10.0.1.182/plugin/prusaslicerthumbnails/thumbnail/Kinder/Monarch.png?20220114081918`
So könnte man es einfach in VIS & Co. anzeigen.

klein0r, 2022-01-15:
Das ist genau da platziert, wo es das Plugin auf der OctoPrint Oberfläche auch hinpackt.
Ich kann ja mal gucken, ob der Printjob auch erweitert wird.

Negalein, 2022-01-15:
> Ich kann ja mal gucken, ob der Printjob auch erweitert wird.

im Plugin kann man `Show thumbnail in state panel.` aktivieren.
Dann scheint es im Printdashboard auch auf.
[image]

klein0r, 2022-01-15:
Muss ich mal einen Printjob starten morgen, dann schaue ich was die API alles hergibt.

Negalein, 2022-01-16:
> Muss ich mal einen Printjob starten morgen, dann schaue ich was die API alles hergibt.

Printjob läuft, aber in den Objekten unter Job taucht das Bild nicht auf

[image]
[image]

klein0r, 2022-01-16:
Wie gesagt, ich muss noch was drucken um zu testen ob mir die API an der Stelle auch die Thumbnail Daten gibt

Negalein, 2022-01-16:
> Wie gesagt, ich muss noch was drucken um zu testen ob mir die API an der Stelle auch die Thumbnail Daten gibt

Ah ok, dann hab ich das falsch verstanden.

Horgul123, 2022-04-25:
Da ich da auch gerade etwas kämpfe: 
Wie komme ich denn jetzt in ioBroker an das Thumbnail? in Octoprint wird es angezeigt, in der Instanz ist die Unterstützung für das Plugin aktiv.

klein0r, 2022-04-25:
> Wie komme ich denn jetzt in ioBroker an das Thumbnail?

Was möchtest Du denn mit dem Thumbnail machen? Zwei Möglichkeiten:

1. Du könntest dir die URL vom Thumbnail aus dem Datenpunkt holen ``files.<name>.thumbnail_url``
2. Du könntest mit den Binärdaten arbeiten, welche im Binary State abgelegt werden ``files.<name>.thumbnail``

https://github.com/klein0r/ioBroker.octoprint/blob/37036240dda4e742e0664999c6851515822a7cd3/main.js#L945-L992

klein0r, 2022-04-25:
Aktuell verwendet das Thumbnail nur eine Hand voll Leute (und ich selbst nichtmal). Das heißt, dass das eventuell noch nicht so easy implementiert ist, wie man sich das ggf. vorstellt 😄 

Stand heute müsstest Du selbst (anhand des file path in der job liste) das richtige file ermitteln und den Pfad holen. Wurde sich aber schon gewünscht, dass die Thumbnail URL dann auch in den aktuellen Job übertragen wird.

Horgul123, 2022-04-25:
Ok... Danke erstmal :)  Würde mir ja den Pfad noch holen, wenn ich wüsste wo... Aber ich kann auch erst mal ohne leben.
Wäre aber schon cool, wenn man in der VIS das Thumbnail anzeigen könnte...

Sonst ein genialer Adapter

Negalein, 2022-04-25:
> Wäre aber schon cool, wenn man in der VIS das Thumbnail anzeigen könnte...

probier dieses Javascript

`// --- Konfiguration ---
const path_octo = 'octoprint.0'; // Instanz zu octoprint
const path_DP_thumbnail = '0_userdata.0.3DDrucker';  // Wo soll der DP liegen - dies muss angelegt sein/werden! Der DP selbst wird automatisch angelegt!
var debug= true; // Log-Meldungen einschalten mit true, ausschalten mit false 
 
//  -- nicht ändern ---
//   Version 1.0.1 2022-01-22
var sel_thumb = path_octo + '.files.*.thumbnail_url';
on({id: path_octo + '.printjob.file.name', change: "ne"}, async function (obj) {
  var octo_thumb='';
  var thumb_file=''; 
  var i = 0;
  if (debug) console.log("Printjob-Name wurde auf >>" + obj.state.val + "<< geändert!");
  if ((obj.state.val)!== null) {
      thumb_file=obj.state.val.replace(".gcode","");
      thumb_file=thumb_file.replace("-","_");
      // thumb_file=thumb_file.substring(0,thumb_file.length)+".thumbnail_url";
      if (thumb_file.length > 0) thumb_file=thumb_file + ".thumbnail_url"; 
      $(sel_thumb).each(function(id) {
          if (id.search(thumb_file) > 0) {
              i++;
              octo_thumb = getState(id).val;
              if (debug) console.log("Printjob-Name >>" + thumb_file + "<< gefunden in " + id + " !");
          };    
      });
      if (debug) {
          console.log(i + " File(s) mit Printjob-Namen " + obj.state.val + " gefunden!");
          if (i>1) console.log("ACHTUNG !!! Mehrere Fundstellen, Datenpunkt " + path_DP_thumbnail + ".thumbnail wurde mit letztem Fund befüllt");
      };    
  } else {
      if (debug) console.log("Printjob-Name ist (null)! Keinen File gefunden!!!");
  }; 
  // DP prüfen ob existent, sonst anlegen
  if (!existsState(path_DP_thumbnail + ".thumbnail")) {
      createState(path_DP_thumbnail + ".thumbnail", octo_thumb,  { name: 'thumbnail', desc: 'full path to thumbnail', type: 'string',   role: '',read: true, write: true });
      if (debug) console.log("Datenpunkt "+ path_DP_thumbnail + ".thumbnail wurde angelegt und mit >>"+ octo_thumb + "<< befüllt!");
  } else {
      setState(path_DP_thumbnail + ".thumbnail", octo_thumb, true);
      if (debug) console.log("Datenpunkt " + path_DP_thumbnail + ".thumbnail wurde mit >>"+ octo_thumb + "<< befüllt!");
  };
  if (debug) console.log("Verarbeitung abgeschlossen!");
}); 
`

Horgul123, 2022-04-25:
Mein Javascript ist nicht so wirklich existent...  Also:
Ich hab das Script angelegt und gestartet. Vorher  das Verzeichnis 0_userdata.0.3DDrucker angelegt.
Darin wird ein DP thumbnail angelegt wenn ich den DP  .printjob.file.name ändere, der thumbnail bleibt aber leer.
Wie aber oben schonmal erwähnt: unter octoprint.0.files sehe ich auch nicht das File, das aktuell gerade gedruckt wird.....

Horgul123, 2022-04-26:
OK... Hab heute noch ein paar Sachen gedruckt und auf einmal gesehen, dass die Files in den Objekten sind. Jetzt geht auch das Script. Danke dafür!

klein0r, 2022-04-28:
Ab Version 3.2.1 ist das nun Standard und die URL des aktuellen files wird nach ``printjob.file.thumbnail_url`` kopiert

Horgul123, 2022-04-29:
Klasse! Vielen Dank dafür. Teste ich später mal

commit 80c1ec79 (2022-01-15): Added plugin Slicer Thumbnails - fixes #65