klein0r/ioBroker.octoprint #67: Zeit für Druckjob

Negalein, 2022-01-19:
### The problem

Es gibt derzeit `octoprint.0.printjob.progress.printtime` (verstrichene Zeit) und `octoprint.0.printjob.progress.printtime_left` (Restzeit).

Cool wäre noch die Gesamtzeit für den Druckjob.
In Octoprint wird sie angezeigt.

Danke

PS: wo ist deine Kaffeekassa? :)

### Version of nodejs

g

### Version of ioBroker js-controller

d

### Version of Adapter

g

### Operating system running ioBroker

z

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

f

joergeli, 2022-01-20:
@Negalein 
Falls du Interesse hast und dich ein weinig mit JavaScript auskennst:

Ich habe mir in ioBroker mit JS aus "verstrichene Zeit" und "Restzeit" einen Datenpunkt gebastelt, wann der Drucker fertig hat ;-)

[image]

Also **um wieviel Uhr** (= Fertigstellung) der Druckjob fertig ist.
(Datum ist mit drin, weil der Druckjob ja auch bis zum nächsten Tag laufen könnte.)

Gesamtzeit könnte ich auch noch stricken.

Kann das Script bei Bedarf gerne im ioBroker-Forum einstellen.

Gruß
Jörg

Negalein, 2022-01-20:
> Gesamtzeit könnte ich auch noch stricken.
> 
> Kann das Script bei Bedarf gerne im ioBroker-Forum einstellen.

Hi Jörg!

Danke, das wäre toll.
Somit könnte @klein0r sich das im Adapter ersparen.

commit 77311a44 (2022-02-23): Calculate date/time when print will be finished - fixes #67
closed 2022-02-23 by klein0r