klein0r/ioBroker.octoprint #69: keine DP mehr in octoprint.0.files für thumbnail, thumbnail_url

sjfm-design, 2022-01-31:
### The problem

gestern aufgefallen beim slicen. seit 3.0.0 werden bei mir keine der genannten DP mehr erzeugt.
in octoprint werden die thumbs aber angezeigt.
am 20. jänner hat das noch funktioniert, wie ich in meinen octoprint.0.files sehen kann.

### Version of nodejs

14.18.2

### Version of ioBroker js-controller

3.3.22

### Version of Adapter

3.3.0

### Operating system running ioBroker

Linux octopi 5.10.63-v7

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

sjfm-design, 2022-02-01:
nach 2 tagen fehler gefunden! in den einstllungen wurden meine plugin einstellungen beim update geändert!
[image]
jetzt wieder die einstllungen, wie vor dem update...

klein0r, 2022-02-18:
Also alles behoben? Ich mach hier mal zu 😄

closed 2022-02-18 by klein0r