klein0r/ioBroker.octoprint #70: Meldung "timeout of 2000ms exceeded"

FreunderNacht, 2022-02-05:
### The problem

Hi,
immer wenn der Octopi eine Verbindung zum Drucker bekomme ich ständig folgende Meldungen im Log:

<html>
<body>
<div class="MuiDrawer-root-178 MuiDrawer-docked-179 jss160 jss161" style="box-sizing: inherit; flex: 0 0 auto; display: flex; transition: width 225ms cubic-bezier(0, 0, 0.2, 1) 0ms; flex-direction: column; width: 180px; color: rgb(255, 255, 255); font-family: &quot;Segoe UI&quot;, Tahoma, Arial, &quot;Courier New&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="MuiPaper-root-34 MuiDrawer-paper-180 jss163 MuiDrawer-paperAnchorLeft-181 MuiDrawer-paperAnchorDockedLeft-185 MuiPaper-elevation0-37" style="box-sizing: inherit; color: rgb(255, 255, 255); transition: box-shadow 300ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; background-color: rgb(18, 18, 18); box-shadow: none; top: 0px; flex: 1 0 auto; height: 1302.67px; display: flex; outline: 0px; z-index: 1200; position: fixed; overflow: hidden auto; flex-direction: column; left: 0px; right: auto; border-right: 1px solid rgba(255, 255, 255, 0.12); width: inherit;"><div class="jss164 jss166" style="box-sizing: inherit; top: 0px; display: flex; padding: 0px 8px; z-index: 2; position: sticky; background: rgb(18, 18, 18); min-height: 64px; align-items: center; justify-content: space-between;"><div class="jss172 jss174 jss173" style="box-sizing: inherit; width: 114.667px; display: flex; margin-top: 5px; margin-bottom: 5px; opacity: 1; transition: opacity 0.3s ease 0s;"><p class="MuiTypography-root-199 jss176 MuiTypography-body1-201" style="box-sizing: inherit; margin: 0px 0px 0px 5px; font-size: 10px; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 400; line-height: 1.5; letter-spacing: 0.00938em; color: rgba(255, 255, 255, 0.37); align-self: center;">.3</p></div><button class="MuiButtonBase-root-85 MuiIconButton-root-77" tabindex="0" type="button" style="box-sizing: inherit; color: rgb(255, 255, 255); border: 0px; cursor: pointer; margin: 0px; display: inline-flex; outline: 0px; padding: 12px; position: relative; align-items: center; user-select: none; border-radius: 50%; vertical-align: middle; justify-content: center; text-decoration: none; background-color: transparent; appearance: none; -webkit-tap-highlight-color: transparent; flex: 0 0 auto; overflow: visible; font-size: 1.5rem; text-align: center; transition: background-color 150ms cubic-bezier(0.4, 0, 0.2, 1) 0ms;"><span class="MuiIconButton-label-84" style="box-sizing: inherit; width: 24px; display: flex; align-items: inherit; justify-content: inherit;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M15.41 7.41L14 6l-6 6 6 6 1.41-1.41L10.83 12z"></path></svg></span><span class="MuiTouchRipple-root-150" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></button></div><ul class="MuiList-root-229 jss167 MuiList-padding-230" style="box-sizing: inherit; margin: 0px; padding: 0px 0px 8px; position: relative; list-style: none; flex: 1 0 auto;"><div draggable="true" data-handler-id="T0" style="box-sizing: inherit; cursor: move; opacity: 1;"><a type="box" href="https://192.168.2.98:8081/#tab-intro" data-handler-id="T0" style="box-sizing: inherit; cursor: move; opacity: 1; color: inherit; text-decoration: none;"><div style="box-sizing: inherit; display: flex;"><div class="MuiButtonBase-root-85 MuiListItem-root-237 MuiListItem-gutters-242 MuiListItem-button-243" tabindex="0" role="button" aria-disabled="false" style="box-sizing: border-box; color: inherit; border: 0px; cursor: pointer; margin: 0px; display: flex; outline: 0px; padding: 8px 16px; position: relative; align-items: center; user-select: none; border-radius: 0px; vertical-align: middle; justify-content: flex-start; text-decoration: none; background-color: transparent; appearance: none; -webkit-tap-highlight-color: transparent; width: 178.667px; text-align: left; transition: background-color 150ms cubic-bezier(0.4, 0, 0.2, 1) 0ms;"><div class="MuiGrid-root-245 jss235 MuiGrid-container-246 MuiGrid-spacing-xs-1-268 MuiGrid-align-items-xs-center-254" style="box-sizing: border-box; width: calc(100% + 8px); display: flex; flex-wrap: nowrap; align-items: center; margin: -4px; height: 40px;"><div class="MuiGrid-root-245 MuiGrid-item-247" style="box-sizing: border-box; margin: 0px; padding: 4px;"><div class="MuiListItemIcon-root-348" style="box-sizing: inherit; color: rgb(255, 255, 255); display: inline-flex; min-width: 0px; flex-shrink: 0;"><span class="MuiBadge-root-106" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;"><svg class="MuiSvgIcon-root-86" focusable="false" viewBox="0 0 24 24" aria-hidden="true"><path d="M4 8h4V4H4v4zm6 12h4v-4h-4v4zm-6 0h4v-4H4v4zm0-6h4v-4H4v4zm6 0h4v-4h-4v4zm6-10v4h4V4h-4zm-6 4h4V4h-4v4zm6 6h4v-4h-4v4zm0 6h4v-4h-4v4z"></path></svg><span class="MuiBadge-badge-107 MuiBadge-anchorOriginTopRightRectangle-112 MuiBadge-colorPrimary-108 MuiBadge-invisible-128" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 195ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; color: rgba(0, 0, 0, 0.87); background-color: rgb(77, 171, 245); top: 0px; right: 0px; transform: scale(0) translate(50%, -50%); transform-origin: 100% 0%;">0</span></span></div></div><div class="MuiGrid-root-245 MuiGrid-item-247" style="box-sizing: border-box; margin: 0px; padding: 4px;"><div class="MuiListItemText-root-350" style="box-sizing: inherit; flex: 1 1 auto; min-width: 0px; margin-top: 4px; margin-bottom: 4px;"><span class="MuiTypography-root-199 MuiListItemText-primary-354 MuiTypography-body1-201 MuiTypography-displayBlock-228" style="box-sizing: inherit; margin: 0px; font-size: 1rem; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 400; line-height: 1.5; letter-spacing: 0.00938em; display: block;"><span class="MuiBadge-root-106" style="box-sizing: inherit; display: inline-flex; position: relative; flex-shrink: 0; vertical-align: middle;">Übersicht<span class="MuiBadge-badge-107 MuiBadge-anchorOriginTopRightRectangle-112 MuiBadge-colorPrimary-108 MuiBadge-invisible-128" style="box-sizing: border-box; height: 20px; display: flex; padding: 0px 6px; z-index: 1; position: absolute; flex-flow: row wrap; font-size: 0.75rem; min-width: 20px; transition: transform 195ms cubic-bezier(0.4, 0, 0.2, 1) 0ms; align-items: center; font-family: Roboto, Helvetica, Arial, sans-serif; font-weight: 500; line-height: 1; place-content: center; border-radius: 10px; color: rgba(0, 0, 0, 0.87); background-color: rgb(77, 171, 245); top: 0px; right: 0px; transform: scale(0) translate(50%, -50%); transform-origin: 100% 0%;">0</span></span></span></div></div></div><span class="MuiTouchRipple-root-150" style="box-sizing: inherit; inset: 0px; z-index: 0; overflow: hidden; position: absolute; border-radius: inherit; pointer-events: none;"></span></div></div></a></div><div draggable="true" data-handler-id="T2" style="box-sizing: inherit; cursor: move; opacity: 1;"><a type="box" href="https://192.168.2.98:8081/#tab-info" data-handler-id="T2" style="box-sizing: inherit; cursor: move; opacity: 1; color: inherit; text-decoration: none;"><div style="box-sizing: inherit; display: flex;"><div class="MuiButtonBase-root-85 MuiListItem-root-237 MuiListItem-gutters-242 MuiListItem-button-243" tabindex="0" role="button" aria-disabled="false" style="box-sizing: border-box; color: inherit; border: 0px; cursor: pointer; margin: 0px; display: flex; outline: 0px; padding: 8px 16px; position: relative; align-items: center; user-select: none; border-radius: 0px; vertical-align: middle; justify-content: flex-start; text-decoration: none; background-color: transparent; appearance: none; -webkit-tap-highlight-color: transparent; width: 178.667px; text-align: left;  [... cut ...]
### Version of Adapter

2.1.0

### Operating system running ioBroker

Ubuntu 20.04 LTS

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

Hier Lof-File des IO Brokers mit dem Adapter im Debug-Mode:
[Log File Io Broker.txt](https://github.com/klein0r/ioBroker.octoprint/files/8008698/Log.File.Io.Broker.txt)

klein0r, 2022-02-05:
Dann stell doch mal den Timeout in den Instanz-Einstellungen höher 😄

FreunderNacht, 2022-02-05:
Hallo,
das habe auch schon versucht (habe ich eben leider nicht erwähnt).

Ich kann die Zeiten nur leider nicht mehr verstellen.

Sobald ich eine Zeit verstelle ist der Speicher Button ausgegraut:
[image]

Dabei ist es egal ob der Adapter läuft oder gestoppt ist.

FreunderNacht, 2022-02-05:
Ich hab die Aktualisierungsintervall schon von Anfang an wie folgt stehen:

[image]

Daran hatte ich seit ich den Adapter installiert hatte, nichts geändert.
Hat auch lange Zeit einwandfrei funktioniert.

klein0r, 2022-02-05:
> Ich kann die Zeiten nur leider nicht mehr verstellen

Das ist leider ein Bug im Admin. Habe dafür schon ein Issue aufgemacht. Am besten mit "hoch runter" Buttons verändern und keine neuen Werte per Tastatur eingeben - dann tritt das Problem nicht auf soweit ich weiß.

FreunderNacht, 2022-02-05:
Ok, mit hoch / runter gehts.

Selbst mit folgenden Einstellungen kommt die Meldung:
[image]

Kann ich noch etwas versuchen?

FreunderNacht, 2022-03-06:
@klein0r 
Thanks for the patch, I just got version 3.1 and installed it. This gives me the setting option I was looking for:

[image]

I have now simply tried 5 seconds.
I still have to try out how far I can go down.
But with the 5 seconds, the problems are already gone.

Thanks a lot

closed 2022-03-06 by FreunderNacht