klein0r/ioBroker.octoprint #74: Zeit wird nicht immer richtig angezeigt

DcFamas, 2022-04-24:
### The problem

Hallo Matthias,
immer wenn ich wie folgt vorgehe, wird die Zeit nicht in den Datenpunkten angezeigt:

- Drucker über Shelly 1PM einschalten (Dauerstrom)
- Drucker geht an
- da ich das Filament wechseln möchte; Preheat PLA/ABS im Menu
- Filament "rausziehen", neues Filament einführen
- Druckjob starten

Dann wird mir in ioBroker der Datenpunkt nur so angezeigt:

[image]

Da ich die Zeit in meinem node.red Flow verwende:

[image]

Wird der Drucker irrtümlicherweise ausgeschalten, obwohl der Druckjob noch läuft. Alle Datenpunkte aktualisieren sich vernünftig, nur die verbleibende Druckzeit leider nicht.
Weißt du wie wir dem auf die Schliche gehen könnten?

Danke dir und Grüße!
DcFamas

### Version of nodejs

latest stable

### Version of ioBroker js-controller

latest stable

### Version of Adapter

v3.2.0

### Operating system running ioBroker

Debian

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

<html><body>

octoprint.0 | 2022-04-24 22:13:32.064 | debug | received  200 response from /plugin/DisplayLayerProgress/values with content:  {"currentFilename":"CE5_Bottom_Right_v01.gcode","fanSpeed":"Off","feedrate":"6000","feedrateG0":"6000","feedrateG1":"1200","height":{"current":"0.12","currentFormatted":"0.1","total":"-","totalFormatted":"-"},"layer":{"averageLayerDuration":"-","averageLayerDurationInSeconds":"-","current":"0","lastLayerDuration":"","lastLayerDurationInSeconds":0,"total":"332"},"print":{"changeFilamentCount":0,"changeFilamentTimeLeft":"-","changeFilamentTimeLeftInSeconds":0,"estimatedChangedFilamentTime":"-","estimatedEndTime":"-","m73progress":"","printerState":"printing","progress":"1","timeLeft":"-","timeLeftInSeconds":"-"}}
-- | -- | -- | --
octoprint.0 | 2022-04-24 22:13:32.059 | debug | received  200 response from /api/printer with content:  {"sd":{"ready":false},"state":{"error":"","flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"paused":false,"pausing":false,"printing":true,"ready":false,"resuming":false,"sdReady":false},"text":"Printing"},"temperature":{"bed":{"actual":50.07,"offset":0,"target":50},"tool0":{"actual":200,"offset":0,"target":200}}}
octoprint.0 | 2022-04-24 22:13:32.048 | debug | received  200 response from /api/job with content:  {"job":{"averagePrintTime":null,"estimatedPrintTime":null,"filament":null,"file":{"date":1650830655,"display":"CE5_Bottom_Right_v01.gcode","name":"CE5_Bottom_Right_v01.gcode","origin":"local","path":"CE5_Bottom_Right_v01.gcode","size":1496373},"lastPrintTime":null,"user":"administrator"},"progress":{"completion":1.8743989633600713,"filepos":28048,"printTime":555,"printTimeLeft":null,"printTimeLeftOrigin":"linear"},"state":"Printing"}
octoprint.0 | 2022-04-24 22:13:32.029 | debug | updatePrinterStatus from: "Printing" -> printerOperational: true, printerPrinting: true
octoprint.0 | 2022-04-24 22:13:32.029 | debug | received  200 response from /api/connection with content:  {"current":{"baudrate":115200,"port":"/dev/ttyUSB0","printerProfile":"_default","state":"Printing"},"options":{"baudratePreference":115200,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":"/dev/ttyUSB0","ports":["/dev/ttyUSB0"],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Ender  5"}]}}
octoprint.0 | 2022-04-24 22:13:31.915 | debug | sending "get" request to "/plugin/DisplayLayerProgress/values" without data
octoprint.0 | 2022-04-24 22:13:31.914 | debug | Starting plugin request
octoprint.0 | 2022-04-24 22:13:31.900 | debug | sending "get" request to "/api/job" without data
octoprint.0 | 2022-04-24 22:13:31.899 | debug | Starting service request
octoprint.0 | 2022-04-24 22:13:31.899 | debug | Plugin "Display Layer Progress" is activated - fetching details
octoprint.0 | 2022-04-24 22:13:31.897 | debug | sending "get" request to "/api/printer" without data
octoprint.0 | 2022-04-24 22:13:31.897 | debug | Starting service request
octoprint.0 | 2022-04-24 22:13:31.895 | debug | sending "get" request to "/api/connection" without data
octoprint.0 | 2022-04-24 22:13:31.895 | debug | Starting service request
octoprint.0 | 2022-04-24 22:13:31.895 | debug | connected to OctoPrint API - online! - status: 200
octoprint.0 | 2022-04-24 22:13:31.894 | debug | received 200 response from /api/version with content: {"api":"0.1","server":"1.7.3","text":"OctoPrint 1.7.3"}
octoprint.0 | 2022-04-24 22:13:31.857 | debug | refreshStateTimeout: re-created refresh timeout (printing): id 3271 - seconds: 15
octoprint.0 | 2022-04-24 22:13:31.856 | debug | sending "get" request to "/api/version" without data
octoprint.0 | 2022-04-24 22:13:31.855 | debug | Starting service request
octoprint.0 | 2022-04-24 22:13:31.855 | debug | refreshState: started from "timeout (printing)"

</body>
</html>

klein0r, 2022-04-25:
Das ist aus meiner Sicht korrekt und muss in deinem Flow abgefangen werden.
Wenn der Druck steht, kann die verbleibende Druckzeit ja nicht angegeben werden. Was soll da stehen? Deswegen liefert OctoPrint an der Stelle null

DcFamas, 2022-04-25:
Hey Matthias,
der Druck steht ja in dem Moment nicht.
Drucker wird eingeschaltet -> Filament wird getauscht -> Druck wird gestartet

Auch das Objekt "completion" wird ja auch verändert. Nur die beiden Werte der verbleibenden Zeit werden nicht korrekt behandelt.

Grüße

klein0r, 2022-04-25:
Das heißt, dass printtimeLeft auf (null) bleibt, obwohl der Filamentwechsel abgeschlossen ist und der Druck schon weiter läuft?

DcFamas, 2022-04-25:
Exakt. Irgendwie kommt Octoprint/der Adapter nicht damit klar, dass ich bevor ich etwas drucke, noch das Menu am Drucker genutzt habe. Wenn ich den Drucker einschalte und direkt anfange zu drucken, gibt es keine Probleme.

Ich nutze aber auch nicht das Menu _Change Filament_ im Druckermenu. Sondern einfach nur _Preheat PLA/ABS_ und ziehe dann, wenn die Nozzle heiß ist, das Material aus dem Drucker. Dann führe ich das neue Filament ein und starte den Druck.
Habe meinen Flow nun auf completion umgestellt, das funktioniert soweit auch wieder.
Generell können wir dem aber gerne auf den Grund gehen.

klein0r, 2022-04-25:
Kannst Du mal zeigen wie das in OctoPrint selbst danach aussieht? Wird es da richtig angezeigt? Der Adapter macht da ja nichts besonderes - ich nehme 1:1 den Wert, welchen mir OctoPrint über die API liefert.

DcFamas, 2022-04-25:
Gerade läuft ein Druck, wenn der vorrüber ist stelle ich das nach. :)

klein0r, 2022-04-25:
Du siehst ja auch in dem Log oben, dass der Wert als ``null`` übergeben wird:

```json
{
    "job": {
        "averagePrintTime": null,
        "estimatedPrintTime": null,
        "filament": null,
        "file": {
            "date": 1650830655,
            "display": "CE5_Bottom_Right_v01.gcode",
            "name": "CE5_Bottom_Right_v01.gcode",
            "origin": "local",
            "path": "CE5_Bottom_Right_v01.gcode",
            "size": 1496373
        },
        "lastPrintTime": null,
        "user": "administrator"
    },
    "progress": {
        "completion": 1.8743989633600713,
        "filepos": 28048,
        "printTime": 555,
        "printTimeLeft": null,
        "printTimeLeftOrigin": "linear"
    },
    "state": "Printing"
}
```

DcFamas, 2022-04-25:
Ja du hast natürlich Recht. Octoprint zeigt auch keine verbleibende Zeit an. Sorry fürs verrückt machen, ich nerve die Leute mal bei Octoprint. 👯

closed 2022-04-25 by DcFamas