klein0r/ioBroker.octoprint #75: this.setForeignBinaryState is not a function

Negalein, 2022-04-29:
### The problem

bekomme haufenweise dieser Meldungen im Log.

`this.setForeignBinaryState is not a function`

### Version of nodejs

14.19.1

### Version of ioBroker js-controller

3.3.22

### Version of Adapter

3.2.1

### Operating system running ioBroker

Ubuntu

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2022-04-29:
Habe vergessen die js-controller dependency hoch zu stellen 😭
Mindestens 4.0.15 nötig

Negalein, 2022-04-29:
> Habe vergessen die js-controller dependency hoch zu stellen 😭 Mindestens 4.0.15 nötig

Danke, hab gleich auf 4.0.23 upgedatet! :)

Negalein, 2022-04-30:
seit JS-Controller 4.0.23 keine Probleme!

klein0r, 2022-04-30:
Ja ich muss noch die dependency erhöhen damit sonst niemand in die Falle läuft

klein0r, 2022-05-01:
Fixed in 3.2.2

closed 2022-05-01 by klein0r