klein0r/ioBroker.octoprint #76: Bug report

Negalein, 2022-05-07:
### The problem

bekomme haufenweise dieser Meldungen.

Octoprint läuft, Drucker ist aus.

`received 400 response from plugin/prusaslicerthumbnails/thumbnail/Mario/Bilskirnir_body_groß.png?20220506175305`

### Version of nodejs

14.19.1

### Version of ioBroker js-controller

4.0.23

### Version of Adapter

3.2.2

### Operating system running ioBroker

Ubuntu

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2022-05-13:
Ich habe mit einer neuen Version angefangen, in welcher sich vieles für die Plugins geändert hat. Noch etwas Geduld bitte.

Negalein, 2022-05-13:
> Ich habe mit einer neuen Version angefangen, in welcher sich vieles für die Plugins geändert hat. Noch etwas Geduld bitte.

Danke

klein0r, 2022-05-19:
Please test with 4.0.0

closed 2022-05-19 by klein0r