klein0r/ioBroker.octoprint #104: PSU Controll Plugin

reptil1990, 2022-09-13:
### The problem

Hey, 

Not a bub but a feature request.

the addidion of the PSU controll plugin would be amazing.

I think that would make it possible to link several outlets to the printer.
And with iObroker the possibility to shut down the printer, ex. in case a smart firealarm goes off, would be great.

Thank you

reptil1990

### Version of nodejs

-

### Version of ioBroker js-controller

-

### Version of Adapter

-

### Operating system running ioBroker

-

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2022-09-13:
Please add a link to the api documentation of the plugin

reptil1990, 2022-09-13:
Here the Plugin: https://github.com/kantlivelong/OctoPrint-PSUControl

I think there is an API under: https://github.com/kantlivelong/OctoPrint-PSUControl/tree/master/octoprint_psucontrol/static/js

Thanks!