klein0r/ioBroker.octoprint #111: Bug report plugin slicer thumbnails erzeugt eine Menge Timeouts in den Logs

inbux, 2022-10-03:
### The problem

Hallo ,

wenn man die thubnails aktiviert hat bekommt man nach einiger Zeit, gibt es sehr viele Timeout in den Logfiles 
 refreshFiles] [plugin slicer thumbnails] error ECONNABORTED from plugin/prusaslicerthumbnails/thumbnail/{_filename_} timeout of 2000ms exceeded
Die Meldungen machen z.Z. etwa 90% aller Logmeldungen aus.

### Version of nodejs

v16.17.0

### Version of ioBroker js-controller

 4.0.23

### Version of Adapter

 4.0.0

### Operating system running ioBroker

linux

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

[iobroker.2022-10-03.log.txt.zip](https://github.com/klein0r/ioBroker.octoprint/files/9699122/iobroker.2022-10-03.log.txt.zip)

klein0r, 2022-10-08:
Mh, warum sollte da Timeouts geben?
Das heißt ja nur, dass die OctoPrint API nicht schnell genug geantwortet hat. Was ist dein Wunsch? Ich könnte einfach den Timeout verlängern - aber ob das so zielführend ist?

klein0r, 2022-10-08:
Passiert das bei allen Dateien? Oder sind die betroffenen besonders groß oder haben sehr große Bilder? Wenn Du die URL zum Thumbnail direkt aufrufst - wie lange dauert es bis zu einer Antwort?

inbux, 2022-10-08:
Ich habe mal ein paar 10 Bilder ausprobiert, die in den Logs einen Timeout hatten.
Die Antwort dauert meist keine 100ms   - die Thumbnails sind relativ klein um 50kB.

Was mir ausgefallen ist:
In den Logs kommen die Timeoutmeldungen ja sehr dicht nacheinander - das sind ja schon so einige Meldungen pro Sekunde.
Vielleicht ist das Abfragen zu schnell, so dass der Webserver irgendwann nicht mehr nachkommt ?
Erst kommt das Timeout ja nur vereinzelt, dann nimmt es zu und irgendwann dann nur noch timeouts.
Kann evtl. auch an der Anzahl der Dateien liegen. Das sind schon knapp 500 Dateien im upload Verzeichnis.

Timeout verlängern, wäre ja schon mal ein Ansatz - vielleicht dynamisch. Genauso das Verlangsamen der Anfragen.
Vielleicht bei zu vielen Timeouts von info auf debug wechseln - oder nur eine info Sammelmeldung, dass es zu Timeouts gekommen ist - damit das Log nicht so voll geschrieben wird.

Ich benutze übrigens einen Rapberry Pi 4 für Octoprint.

klein0r, 2022-10-08:
Verstehe - mit so vielen Dateien habe ich es noch nicht getestet. Dann baue ich das mal auf synchron um, dass eine Datei nach der anderen abgefragt wird. Aber das kann dann schon sehr lange dauern.

500 * 100ms sind halt trotzdem 50 Sekunden. Ich sorge mal dafür, dass jedes Bild nur 1x geholt wird.