klein0r/ioBroker.octoprint #113: Command Befehle

TomsEleven, 2022-11-06:
### The problem

Hallo Matthias,

vielen Dank für die Programmierung dieses Adapters. Funktioniert bestens.

In Deinem Video beschreibst Du, dass man in den Command Objekten Werte setzen kann & falls diese nicht existieren man eine Liste der erlaubten Befehle im Log erhält. Ich hab Dir unten mal den Ausschnitt des Logs reinkopiert. Bei mir kommt nur: Choose one of:   und dann nichts mehr.

In meinem Fall geht es mir um einen Befehl mit dem ich meinen Raspi runterfahren kann. Mit shutdown funktionierts offensichtlich nicht. Hast Du evtl. eine Liste mit allen möglichen Werten?

Vielen Dank & viele Grüße
Tom

### Version of nodejs

14.20.1

### Version of ioBroker js-controller

4.0.15

### Version of Adapter

4.0.0.

### Operating system running ioBroker

Linux VM

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

2022-11-04 15:33:54.737 - info: host.IOBroker "system.adapter.octoprint.0" enabled--
2022-11-04 15:33:55.024 - info: host.IOBroker instance system.adapter.octoprint.0 started with pid 4703
2022-11-04 15:33:56.600 - info: octoprint.0 (4703) starting. Version 4.0.0 in /opt/iobroker/node_modules/iobroker.octoprint, node: v14.20.1, js-controller: 4.0.15
2022-11-04 15:36:19.341 - error: octoprint.0 (4703) system command not allowed: ghg. Choose one of:
2022-11-04 15:37:09.829 - info: octoprint.0 (4703) error ECONNABORTED from /api/connection: timeout of 2000ms exceeded
2022-11-04 15:38:40.888 - error: octoprint.0 (4703) system command not allowed: shutdown. Choose one of:
2022-11-04 15:47:38.346 - info: host.IOBroker "system.adapter.octoprint.0" disabled

klein0r, 2022-11-06:
> Choose one of: und dann nichts mehr.

Dann zeig doch mal, welche Befehle Du in OctoPrint konfiguriert hast und welche dort auftauchen müssten

TomsEleven, 2022-11-06:
Super. Vielen Dank. Das war der entscheidende Hinweis. Hab Shutdown konfiguriert:
https://docs.octoprint.org/en/master/configuration/config_yaml.html?highlight=shutdown#system

Jetzt funktionierts :-)

Viele Grüße
Tom

closed 2022-11-10 by klein0r