klein0r/ioBroker.octoprint #129: High CPU usage during sync

Kaskadeur111, 2023-03-17:
### The problem

Dont know really if its a bug, because the adapter runs fine otherwise.

I am running Adapter Version v4.0 and tested v4.1 because i noticed frequently CPU spikes on my Virtual Machine where iobroker runs.

They are comming from the octoprint adapter. Everytime it syncs to my Octoprint Raspi it uses a full Core (i5 3.5GHz) for about 3 seconds. Also i can see there are many input Operations during sync (too much?)

[image]

It is no difference if the octoprint Raspi is online or offline (powerless). The only difference is, the time intervals where the spikes happen are not the same (because i set different sync intervals for connected/not connected.

No other adapter causes this much load (if it would be a raspi running iobroker i could imagine it needs some cpu power to do the things it needs to do, but not with a i5 and 16GB of ram on an SSD)

I didnt notice this spikes in the past (my sync intervals where set to 5 seconds). For testing purposes i set them to 30 seconds and 10 seconds when connected.

Also i dont have checked the Thumbnails from Slicer

[image]

i will test if this happens in v3.x also.

### Version of nodejs

16.19.0

### Version of ioBroker js-controller

4.0.24

### Version of Adapter

4.0.0 and 4.1.0

### Operating system running ioBroker

Linux Debian latest Version

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2023-03-17:
> iobroker.current.log (Please share the logs in debug mode)

Kaskadeur111, 2023-03-17:
Thanks for the hint with the logs! I think i got the Problem... on every sync it syncs the whole file List from the octoprint history!

[image]

Maybe it would be a nice option to diable the file sync (if not needed at all) or to configure a Time Range for the file sync (all files younger than 6 Months or something like this).

I can delete all my files in octoprint, but i dont really want to do this :-)

Kaskadeur111, 2023-03-17:
By the way, the screenshot is with adpater version 3.0 (tested this right now and can confirm the same behaviour) :-)