klein0r/ioBroker.octoprint #139: Missing thumbnail_url

Horgul123, 2023-05-24:
### The problem

I used the thumbnail_url to get the image for my VIS.
I noticed, that it's no longer working. 
I'm slicing with Cura 5.3.1, in postprocessing "Create Thumbnail" is active. In Octoprint 1..8.7 I can see the thumbnails.
After checking the adapter (4.1.0) I saw, that the thumbnail_url is missing.

I have no idea since when this issue exists, 
In the iobroker forum I saw that other people have this problem, too. But no solution was found

### Version of nodejs

v18.15.0

### Version of ioBroker js-controller

4.0.24

### Version of Adapter

4.1.0

### Operating system running ioBroker

Debian 5.10.149-2 (2022-10-21) x86_64

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

No log available at the moment (long print in progress). I needed I'll create one

klein0r, 2023-05-24:
Please share a log in debug log mode

Horgul123, 2023-05-24:
Just started a print:

`<html>
<body>

octoprint.0 | 2023-05-24 09:03:51.819 | debug | [plugin slicer thumbnails] found current file: files.local_20x20x15_mm_hollow_cube_aw
-- | -- | -- | --
octoprint.0 | 2023-05-24 09:03:51.802 | debug | received 200 response from /plugin/DisplayLayerProgress/values with content: {"currentFilename":"20x20x15_mm_hollow_cube.aw.gcode","fanSpeed":"-","feedrate":"-","feedrateG0":"-","feedrateG1":"-","height":{"current":"-","currentFormatted":"-","total":"20.200000762939453","totalFormatted":"20.2"},"layer":{"averageLayerDuration":"-","averageLayerDurationInSeconds":"-","current":"-","lastLayerDuration":"-","lastLayerDurationInSeconds":"-","total":"99"},"print":{"changeFilamentCount":0,"changeFilamentTimeLeft":"-","changeFilamentTimeLeftInSeconds":0,"estimatedChangedFilamentTime":"-","estimatedEndTime":"09:27","m73progress":"","printerState":"printing","progress":"0","timeLeft":"23m","timeLeftInSeconds":1424}}
octoprint.0 | 2023-05-24 09:03:51.799 | debug | [plugin slicer thumbnails] trying to find current print job thumbnail url
octoprint.0 | 2023-05-24 09:03:51.797 | debug | received 200 response from /api/job with content: {"job":{"averagePrintTime":null,"estimatedPrintTime":1424.7229110976868,"filament":{"tool0":{"length":1144.0841064155102,"volume":2.7518447650997917}},"file":{"date":1684908876,"display":"20x20x15_mm_hollow_cube.aw.gcode","name":"20x20x15_mm_hollow_cube.aw.gcode","origin":"local","path":"20x20x15_mm_hollow_cube.aw.gcode","size":297646},"lastPrintTime":null,"user":"octoprint"},"progress":{"completion":9.653413786847462,"filepos":28733,"printTime":0,"printTimeLeft":1424,"printTimeLeftOrigin":"genius"},"state":"Printing"}
octoprint.0 | 2023-05-24 09:03:51.794 | debug | received 200 response from /api/printer with content: {"sd":{"ready":false},"state":{"error":"","flags":{"cancelling":false,"closedOrError":false,"error":false,"finishing":false,"operational":true,"paused":false,"pausing":false,"printing":true,"ready":false,"resuming":false,"sdReady":false},"text":"Printing"},"temperature":{"bed":{"actual":50.01,"offset":0,"target":60},"tool0":{"actual":25.59,"offset":0,"target":0}}}
octoprint.0 | 2023-05-24 09:03:51.790 | debug | updatePrinterStatus from: "Printing" -> printerOperational: true, printerPrinting: true
octoprint.0 | 2023-05-24 09:03:51.789 | debug | received 200 response from /api/connection with content: {"current":{"baudrate":250000,"port":"/dev/ttyUSB0","printerProfile":"_default","state":"Printing"},"options":{"baudratePreference":null,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":null,"ports":["/dev/ttyS0","/dev/ttyS1","/dev/ttyS2","/dev/ttyS3","/dev/ttyUSB0"],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Ender 3 S1"}]}}
octoprint.0 | 2023-05-24 09:03:51.785 | debug | sending "get" request to "/plugin/DisplayLayerProgress/values" without data
octoprint.0 | 2023-05-24 09:03:51.785 | debug | [buildPluginRequest] starting plugin request
octoprint.0 | 2023-05-24 09:03:51.777 | debug | sending "get" request to "/api/job" without data
octoprint.0 | 2023-05-24 09:03:51.777 | debug | [buildServiceRequest] starting service request
octoprint.0 | 2023-05-24 09:03:51.777 | debug | [plugin display layer progress] plugin activated - fetching details
octoprint.0 | 2023-05-24 09:03:51.776 | debug | sending "get" request to "/api/printer" without data
octoprint.0 | 2023-05-24 09:03:51.776 | debug | [buildServiceRequest] starting service request
octoprint.0 | 2023-05-24 09:03:51.775 | debug | sending "get" request to "/api/connection" without data
octoprint.0 | 2023-05-24 09:03:51.775 | debug | [buildServiceRequest] starting service request
octoprint.0 | 2023-05-24 09:03:51.775 | debug | connected to OctoPrint API - online! - status: 200
octoprint.0 | 2023-05-24 09:03:51.774 | debug | received 200 response from /api/version with content: {"api":"0.1","server":"1.8.7","text":"OctoPrint 1.8.7"}
octoprint.0 | 2023-05-24 09:03:51.769 | debug | refreshStateTimeout: re-created refresh timeout (printing): id 10360 - seconds: 10
octoprint.0 | 2023-05-24 09:03:51.767 | debug | sending "get" request to "/api/version" without data
octoprint.0 | 2023-05-24 09:03:51.767 | debug | [buildServiceRequest] starting service request
octoprint.0 | 2023-05-24 09:03:51.767 | debug | refreshState: started from "timeout (default)"
octoprint.0 | 2023-05-24 09:02:51.799 | debug | received 200 response from /plugin/DisplayLayerProgress/values with content: {"currentFilename":"20x20x15_mm_hollow_cube.aw.gcode","fanSpeed":"-","feedrate":"-","feedrateG0":"-","feedrateG1":"-","height":{"current":"-","currentFormatted":"-","total":"20.200000762939453","totalFormatted":"20.2"},"layer":{"averageLayerDuration":"-","averageLayerDurationInSeconds":"-","current":"-","lastLayerDuration":"-","lastLayerDurationInSeconds":"-","total":"99"},"print":{"changeFilamentCount":0,"changeFilamentTimeLeft":"-","changeFilamentTimeLeftInSeconds":0,"estimatedChangedFilamentTime":"-","estimatedEndTime":"09:26","m73progress":"","printerState":"printing","progress":"0","timeLeft":"23m","timeLeftInSeconds":1424}}
octoprint.0 | 2023-05-24 09:02:51.795 | debug | (system/commands) registered commands: core/restart, core/restart_safe
octoprint.0 | 2023-05-24 09:02:51.794 | debug | received 200 response from /api/system/commands with content: {"core":[{"action":"restart","confirm":"<strong>You are about to restart the OctoPrint server.</strong></p>< [... cut ...]
octoprint.0 | 2023-05-24 09:02:51.774 | debug | received 200 response from /api/version with content: {"api":"0.1","server":"1.8.7","text":"OctoPrint 1.8.7"}
octoprint.0 | 2023-05-24 09:02:51.766 | debug | refreshStateTimeout: re-created refresh timeout (default): id 9819 - seconds: 60
octoprint.0 | 2023-05-24 09:02:51.764 | debug | sending "get" request to "/api/version" without data
octoprint.0 | 2023-05-24 09:02:51.764 | debug | [buildServiceRequest] starting service request
octoprint.0 | 2023-05-24 09:02:51.763 | debug | refreshState: started from "timeout (default)"

</body>
</html>`

klein0r, 2023-05-24:
```
[plugin slicer thumbnails] found current file: files.local_20x20x15_mm_hollow_cube_aw
```

Normally that should fill the thumbnail url state. But that state is empty?

Horgul123, 2023-05-24:
Now it's strange: The URL was missing during the last 3 prints. This time the URL is filled. 
The last prints were a little bigger, this is just a small 20x20mm test cube.... And now the URL is filled.
I didn't chance anything in the setup of the tools. 

I think I'll check it during the next prints again.... Strange...