klein0r/ioBroker.octoprint #160: octoprint has an invalid jsonConfig:

Brainbug01, 2023-10-30:
### The problem

Admin v6.12.1 meldet beim Adapter öffnen:

```
octoprint has an invalid jsonConfig: 
[{
"instancePath":"",
"schemaPath":"#/additionalProperties",
"keyword":"additionalProperties",
"params":{"additionalProperty":"iconPosition"},
"message":"must NOT have additional properties"
}]
```

### Version of nodejs

 v18.18.2

### Version of ioBroker js-controller

 5.0.15

### Version of Adapter

v5.1.0

### Operating system running ioBroker

Linux

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

klein0r, 2023-10-30:
https://github.com/ioBroker/adapter-react-v5/issues/197

closed 2023-10-30 by klein0r

mcm1957, 2023-10-30:
Hmmm
Wenn das angeblich schon gefixed ist, dann dürfte der Fehler eigentlich nicht mehr auftreten.
Lt. foxriver prüft der admin gegen das ONLINE schema.

klein0r, 2023-10-31:
Naja, da liegen nur ein paar Stunden zwischen Issue und Fix. Eventuell hat sich das auch Überschnitten?

https://github.com/ioBroker/ioBroker.admin/blob/41225cdb7132a2081ab8011ca363613ce8294a90/lib/web.js#L362C11-L368

Anscheinend wird das wirklich bei jedem Öffnen der Konfiguration durchgeführt 😮 

https://github.com/ioBroker/ioBroker.admin/blob/41225cdb7132a2081ab8011ca363613ce8294a90/src/src/components/JsonConfig.tsx#L614C1-L618

mcm1957, 2023-10-31:
Ja, der Check wird bei jedem Öffnen der Config ausgeführt. Aber so oft wird die Config ja wohl nicht geöffnet :-)
Der obige Fehler ist gefixed,

Aber es gibt einen weiteren Fehler bei Multilanguage. Hab dazu issue erstellt:
https://github.com/ioBroker/adapter-react-v5/issues/207

klein0r, 2023-10-31:
> Ja, der Check wird bei jedem Öffnen der Config ausgeführt

Naja, ein Cache wäre schon nett.
Das Repository wird ja z.B. auch nicht jedes Mal geladen, wenn man auf die Adapter-Liste geht.

mcm1957, 2023-10-31:
Im Prinzip geb ich dir recht, Derzeit (mit fehlerhaftem Schema) ist no cache sicher sinnvoll. Mittelfristig macht cache ev. durchaus Sinn. Erstell ggF mal einen Issue (oder gleich PR :-) )

Brainbug01, 2023-12-02:
Die warn Meldung ist noch da, liegt das an mir?

klein0r, 2023-12-02:
> Die warn Meldung ist noch da, liegt das an mir?

Mittlerweile ist iconPosition eigentlich enthalten und die Validierung sollte klappen

Brainbug01, 2023-12-02:
Wie gesagt ist es bei mir immer noch da. Ich bin im Beta unterwegs und habe alle Adapter ( außer sayit ) auf aktuellem Stand.

klein0r, 2023-12-02:
Danke, ich nutze das Attribut auf einem anderen Element als im Schema definiert ist. Habe es nochmal addressiert:

https://github.com/ioBroker/adapter-react-v5/issues/197#issuecomment-1837146845

mcm1957, 2024-01-18:
@klein0r 
Issue sollte reopend werden.

https://github.com/klein0r/ioBroker.octoprint/blob/12f9b463dc62b1d537cc4d10be7f8203282395d6/admin/jsonConfig.json#L4

Attribut iconPosition sollte lt. schema.md nicht auf TOP Ebene sondern innerhalb PANEL / TAB angegeben werden - dort wo auch das icon angegeben wird. Zumindest les ich die Doku so.

klein0r, 2024-01-18:
@mcm1957 see https://github.com/ioBroker/adapter-react-v5/issues/197

mcm1957, 2024-01-18:
@klein0r 
Du hast wieder mal vollkommen recht. Ich hab die Spezifikation falsch gelesen. Da scheint wirklich das SChem flsch zu sein,
Nur ich find dazu kein offenen SCHEMA Issue. Das referenziert Issue ist ja schon zu ...

Ich stells nochmal mit den aktuellen Meldungen ein

klein0r, 2024-01-18:
Ich bekomme überall Issues zu dem Thema und mache die immer zu, weil ja der Admin-Adapter die jsonConfig aufgebaut und das Schema dazu in adapter-react-v5 liegt. Gäbe es hier Handlungsbedarf von meiner Seite, hätte ich das natürlich direkt umgesetzt.

mcm1957, 2024-01-18:
Ja nur schau bitte ob ein passendes Issue in react-v5 existiert und verlink das. Ist für die User verständlicher. Für die iconPosition hab ich mal ein (neues) Issue angelegt.

klein0r, 2024-01-18:
@mcm1957 Genau das mache ich doch? Siehe erste Antwort hier

mcm1957, 2024-01-18:
Passt schon. DANKE jedenalls
Hab Moritz getriggert. Er schaut wann er Zeit freischaufeln kann um die offenen Schmea Dinge mal durchzugehen.

klein0r, 2024-01-18:
Die Validation wäre irgendwie besser im adapter-checker aufgehoben als sie jedem User einzeln als Warning vor die Füße zu werfen...