klein0r/ioBroker.octoprint #182: second instance on port 5001 api not connected

icemanmw, 2024-01-22:
### The problem

i have two instances on iobroker for octoprint. Since a couple of days i get no api connection for the instance with the port 5001 (on 80 it is running , aso if i switch the port between the instances)
Also same issue with a downgrade from 5.1.0 to 4.0.1 

LOG-file - instance octoprint.0 is with port 5001  / octoprint.2 is with port 80

### Version of nodejs

 v20.11.0

### Version of ioBroker js-controller

5.0.17

### Version of Adapter

5.1.0

### Operating system running ioBroker

Proxmox ioBroker LXC

### Checklist of files to include below

- [X] iobroker.current.log (Please share the logs in debug mode)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)

### Additional information & file uploads

<html>
<body>

2024-01-22 22:05:30.740 - info: host.iobroker stopInstance system.adapter.octoprint.0 (force=false, process=true)
--
2024-01-22 22:05:30.743 - info: octoprint.0 (1746503) Got terminate signal TERMINATE_YOURSELF
2024-01-22 22:05:30.743 - info: octoprint.0 (1746503) terminating
2024-01-22 22:05:30.744 - info: octoprint.0 (1746503) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2024-01-22 22:05:30.863 - info: host.iobroker stopInstance system.adapter.octoprint.0 send kill signal
2024-01-22 22:05:31.245 - info: octoprint.0 (1746503) terminating
2024-01-22 22:05:31.304 - info: host.iobroker instance system.adapter.octoprint.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2024-01-22 22:05:33.533 - warn: admin.1 (1742244) octoprint has an invalid jsonConfig: [{"instancePath":"","schemaPath":"#/additionalProperties","keyword":"additionalProperties","params":{"additionalProperty":"iconPosition"},"message":"must NOT have additional properties"}]
2024-01-22 22:05:33.922 - info: host.iobroker instance system.adapter.octoprint.0 started with pid 1751048
2024-01-22 22:05:34.838 - info: octoprint.0 (1751048) starting. Version 5.1.0 (non-npm: klein0r/ioBroker.octoprint) in /opt/iobroker/node_modules/iobroker.octoprint, node: v20.11.0, js-controller: 5.0.17
2024-01-22 22:05:35.445 - info: octoprint.0 (1751048) error ECONNRESET from /api/connection: socket hang up
2024-01-22 22:05:45.443 - info: octoprint.0 (1751048) error ECONNRESET from /api/connection: socket hang up
2024-01-22 22:06:45.489 - info: octoprint.0 (1751048) error ECONNRESET from /api/connection: socket hang up
2024-01-22 22:07:04.520 - info: host.iobroker stopInstance system.adapter.octoprint.1 (force=false, process=true)
2024-01-22 22:07:04.522 - info: octoprint.1 (1745600) Got terminate signal TERMINATE_YOURSELF
2024-01-22 22:07:04.523 - info: octoprint.1 (1745600) terminating
2024-01-22 22:07:04.523 - info: octoprint.1 (1745600) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2024-01-22 22:07:04.567 - info: host.iobroker stopInstance system.adapter.octoprint.1 send kill signal
2024-01-22 22:07:05.024 - info: octoprint.1 (1745600) terminating
2024-01-22 22:07:05.334 - error: octoprint.1 (1745600) Cannot check object existence of "octoprint.1.files.local_AI3M_001100_Klemm_Stoppel_OD4.name": Connection is closed.
2024-01-22 22:07:05.410 - info: host.iobroker instance system.adapter.octoprint.1 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2024-01-22 22:07:07.635 - info: host.iobroker instance system.adapter.octoprint.1 started with pid 1751279
2024-01-22 22:07:08.732 - info: octoprint.1 (1751279) starting. Version 5.1.0 (non-npm: klein0r/ioBroker.octoprint) in /opt/iobroker/node_modules/iobroker.octoprint, node: v20.11.0, js-controller: 5.0.17
2024-01-22 22:07:45.443 - info: octoprint.0 (1751048) error ECONNRESET from /api/connection: socket hang up

</body>
</html>

icemanmw, 2024-01-28:
i found a solution for the port problem. i installed octoprint / octopie new and used the script from https://github.com/paukstelis/octoprint_deploy.  Now i can use both instance on port 80.

icemanmw, 2024-02-03:
Problem is still there. I guess the problem ist the webserver location. in my case it is 192.168.1.x/Vyper and 192.168.1.x/MegaX 
Log still shows "error ECONNRESET from /api/connection: socket hang up"

klein0r, 2024-02-05:
> Version 5.1.0 (non-npm: klein0r/ioBroker.octoprint)

Don't install adapters from GitHub... how shoud I know what is installed?

icemanmw, 2024-02-06:
Hello, you are right , i didn´t recogniced that i installed during the last days a github version whiel i was testing the . 
Now i deinstalled the Github Version and reinstalled the latest official version (i hope so), but still the same issue.

2024-02-06 16:48:35.797 - info: octoprint.1 (2726706) starting. Version 5.1.0 in /opt/iobroker/node_modules/iobroker.octoprint, node: v20.11.0, js-controller: 5.0.17
2024-02-06 16:48:36.024 - info: octoprint.1 (2726706) error ECONNRESET from /api/connection: socket hang up
2024-02-06 16:48:39.866 - info: javascript.0 (312337) script.js.OctoPrint-API.DisplayLayerProgress: request: http://192.168.1.83:5001/plugin/DisplayLayerProgress/values?apikey=9724........xxxxx

klein0r, 2024-02-07:
> error ECONNRESET from /api/connection: socket hang up

But that sounds like a problem on the other side (connection has been closed by octoprint).

icemanmw, 2024-02-07:
It seems that the path is the problem.
I have in parallel the App octopod and the home assistant instance running. In both cases I can point to http://192.168.1.83/Vyper and it woks . 
In the ioBroker.octoprint instance,  I just can point to the  IP without the /vyper.
Maybe that is the reason … I don’t know.